nvdla/hw #94: When I try to generate nv_small, it goes wrong.

CrazyBingo, 2018-02-03:
When I try to generate nv_small, it goes wrong.   It exiting while the project.h does nor exists., and stoped itself.

So can the NVDLA now support nv_small now? TKs.

JunningWu, 2018-02-05:
No, just now.

jwise, 2018-02-06:
@JunningWu is correct.  nv_small is expected in a future release.

teabun, 2018-02-08:
Is there an estimate timeline when nv_small will be available? Some clarity on this is helpful, thanks!

CrazyBingo, 2018-02-08:
@teabun So the same to you, the reason why we want nv_small is the area of nv_large is big

CrazyBingo, 2018-02-08:
@teabun 
sorry I do not know, I'm not the member of NVDIA, but a follower of nvdla as you.
I just find out that we can not generate nv_small code now.

Which step have you reached?

teabun, 2018-02-09:
My team just started explorating nv_large as we are aware nv_small isn't ready,  but from the milestone it seems to suggest end Q4 '17. Just wish to get some visibility for planning purpose.
Thanks!

CrazyBingo, 2018-02-09:
Me too，and we aware thar nv_full can not put in MPSOC， or the performance of nv_small is not enough

megalomanaa, 2018-02-19:
I am also looking forward to the release, but then, is this true that any different MAC size other than nv_full is not available yet?

CrazyBingo, 2018-02-25:
May be，wait together ……

jwise, 2018-04-19:
nv_small is now released in the master branch.

closed 2018-04-19 by jwise