nvdla/hw #96: about DRAM cycles in NVDLA_OpenSource_Performance.xlsx

HKLee2040, 2018-02-07:
There two major cycle counts are used to estimate the performance, one is MAC cycles, and the other is DRAM cycles.
The DRAM cycles in the spreadsheet comes from DRAM traffic, the DRAM traffic comes from input data size, output data size, and weight data size.
If C is multiples of 16, stride = 1, conv 
input data size = C x H x W
output data size = A x B x K
weight data size = C x R x S x K
Is it too optimistic to estimate the data movement?

HKLee2040, 2018-02-09:
The formula used in Alexnet is different from Googlenet or Resnet.
I think Alexnet's formula is too simple.  If the convolution buffer is small, e.g., 64 or 32 kbytes, the DRAM cycles may be incorrect.

giranntu, 2018-02-12:
The first question is: what math precision is required? Training is usually done at 32-bit floating point, but the resulting networks can often be reduced to 8-bit integers without significant loss of inference quality; however, in some cases 16-bit integer or float may be desirable.
 
After precision, the next two critical parameters for performance and area are the number of multiply-accumulates (MACs) and memory bandwidth. When configuring the NVDLA, these should be carefully considered. Processing happens layer-by-layer and performance estimation is best done layer-by-layer as well. For each layer, either MACs or memory traffic will be the bottleneck.
 
The number of MACs is relatively easy to determine. For example, for a convolutional layer, assume the input and output resolution is WxH, and the number of input and output features is C and K respectively, and the convolutional kernel is RxS coefficients. 
The total number of MAC operations is: MACops = W * H * C * K * R * S . 
If the number of MAC hardware units in the DLA is MACunits, then it will take at least MACops / MACunits clock cycles to finish the operations.
 
Bandwidth calculations may be harder. 
In the same example as above, the ideal bandwidth for the input is C * H * W , 
for the output it is W * H * K, 
and for the weights it is C * K * R * S . 
The minimum number of cycles for memory traffic is the sum of these divided by the number of samples that can be read or written per cycle. However, if the convolutional buffer is too small to hold the input or weights, multiple passes are required. For example, consider a layer with 400KB of weights and 10MB of input size. Ideally, the input memory traffic is 10MB. However, assume the convolutional buffer can only hold 100KB of weight data. In that case, the best solution is to split the calculation in four steps, each handling one quarter of the filters and calculating one quarter of the output. The problem with this is that each step needs the full input, So rather than reading the input once, it has to be read four times and the bandwidth for the input increases from the ideal 10MB to 40MB. This effect is important to consider when choosing the convolutional buffer size. Note that the calculation above disregards any extra buffer space needed for double buffering and latency hiding.

HKLee2040, 2018-02-14:
I think at least the behavior should be nearly correct in the spreadsheet.

Here is another example:
In alexent, weight data size item will check whether Winograd is used or not, but the "Min weight data size" doesn't.  Is it correct "behavior"?

ratnayak, 2018-02-23:
i have a related question and that is regarding frames per sec.
both NVDLA primer (http://nvdla.org/primer.html) and NVDLA_OpenSource_Performance.xlsx gives performance numbers for ResNet-50. but the FPS numbers are different.
primer: 269 FPS
NVDLA_OpenSource_Performance: 140 FPS
it looks like most of the settings--clk=1GHs, #MACS=2048, precision=int8--are the same for both cases. 
whats the reason(s) for the almost 2x diff?

HKLee2040, 2018-02-24:
@ratnayak  If you change the item "Primary DBB BW Limitation (GB/s)" from 10GB/s to 20GB/s, the result will be very close to 269 fps.

ratnayak, 2018-03-02:
Thanks HKLee. that explains.

jwise, 2018-03-03:
Looks like most of this got answered.  Is there anything here still open?