nvdla/hw #99: Apprxomation error of non-linear function in SDP

Sonansu04, 2018-02-23:
NVDLA supports different precision FP16/INT16/INT8, and uses LUT to implement non-linear functions.  The DLA introduction document illustrates 127/65 entries for two LUTs.  Is there any error constraints for tanh(x) and sigmoid(x) such as ulp (unit in least precision) or maximum approximate error?

jwise, 2018-03-03:
I'm not sure.  I think this analysis exists somewhere, but I'm not sure where.  I'll see what I can find for you...

fanqifei, 2018-03-03:
Hi @Sonansu04 , we performed precision study and compared the loss rate with caffe. In caffe FP64 tanh(x) and sigmod(x) are used, LUTs are not used.
The difference of loss rate is acceptable. 
There is no error constraint on the result of LUT although we can compute the max error.

Sonansu04, 2018-03-04:
Thank you for the answer. I still have some questions about the size of LUTs and the domain of input.
1. It uses 127/65 entries for two LUTs, so the size of LUTs is 2032 bits(127x16), 1040 bits(65x16) respectively.
2. Is there any upper bound/lower bound for the domain of input? For example, it outputs a constant value for sigmoid(x), when x is larger than 18 (upper bound).

kenter83, 2018-03-05:
Hi, @Sonansu04 , 
     Regarding to your questions:
     #1, You are right, but there're 257 entries for the first LUT, so, it should be 4112 bits (257x16);
     #2, The upper/lower bound of a LUT input is depending on the LUT setting. From hardware perspective, the LUT input is INT32/FP32 and output is INT32/FP32 for INT/FP pipeline respectively thus for the best LUT setting, you can get always get different output. There's a constrain for LUT that the LUT_END - LUT_START must be power of 2 thus for sigmoid, an legal LUT config will cover -32~32, thus there'll be some dynamic range loss for LUT input.
      There're 2 chapters in NVDLA IAS: "LUT programming" and "Limitation.LUT", you can refer to those chapters to get detailed description after IAS is released. @jwise might be able to comment the IAS release plan.

jwise, 2018-03-05:
I'll hope to have some more info on that for you by tomorrow.