nvdla/hw #102: When I run "./tools/bin/tmake -build verif_sim"， it finally goes wrong

CrazyBingo, 2018-02-26:
First, I have correctly setup SystemC/VCS/Phython etc， and have successfully run  ‘/tools/bin/tmake -build vmod’. But When I next run "./tools/bin/tmake -build verif_sim"， it can compile successfully, but there is sth. wrong at the last, the log is like this:

Note-[RT_BS] Larger stack needed
  Note: Bumping stack limit from 8192 to 8195 Kbytes.

Error: Message for 'CMD-015' has occurred (ERR-3)
Error: Message for 'CMD-015' has occurred (ERR-3)
Command line: ./simv -l test.log +read_reg_poll_retries=10 +continue_on_fail +continue_on_fail +input_file=/media/D/NVDLA_20180131/hw-nvdlav1/verif/sim/sanity0/input.txn +input_dir=/media/D/NVDLA_20180131/hw-nvdlav1/verif/sim/sanity0

--- Stack trace follows:

Dumping VCS Annotated Stack:
No context available
make: *** [../traces/traceplayer/sanity0/test.log] Error 1
Makefile:536: recipe for target '../traces/traceplayer/sanity0/test.log' failed
make: Leaving directory '/media/D/NVDLA_20180131/hw-nvdlav1/verif/sim'
logfile: outdir/build.log
==============================================
==================BUILD PASS==================
==============================================

Why it goes wrong finally, and if I should concerned about the failure?

CrazyBingo, 2018-02-26:
the I cd .verif/sim, and make sanity or make check, it failded agin and tells me : 

Checking /media/D/NVDLA_20180131/hw-nvdlav1/verif/sim/sanity0/test.log 
/media/D/NVDLA_20180131/hw-nvdlav1/verif/synth_tb/sim_scripts/checktest_synthtb.pl sanity0 test.log; 
checktest : NOTRUN : sanity0                                  : no log found : please kick off with make run TESTDIR=sanity0 

When I exaime the verif/sim dir, there is only sanity0, no sanity1-3 etc., and there is no synth_tb/.. dir.

CrazyBingo, 2018-02-26:
The problem of stack limitation is resolve by add parameter to etc/security/ulimit.conf, and restart the computer, the parameter is like this:
soft noproc 10240
hard noproc 10240
soft nofile 10240
hard nifile 10240

while run ./tools/bin/tmake -build verif_sim again ,the stack limit is OK, but the result is faild while generate sanity0 agin. the logs is like this:
rror: Message for 'CMD-015' has occurred (ERR-3)
Error: Message for 'CMD-015' has occurred (ERR-3)
Command line: ./simv -l test.log +read_reg_poll_retries=10 +continue_on_fail +continue_on_fail +input_file=/media/D/NVDLA_20180131/hw-nvdlav1_v1/verif/sim/sanity0/input.txn +input_dir=/media/D/NVDLA_20180131/hw-nvdlav1_v1/verif/sim/sanity0

--- Stack trace follows:

Dumping VCS Annotated Stack:
No context available
make: *** [../traces/traceplayer/sanity0/test.log] Error 1
Makefile:538: recipe for target '../traces/traceplayer/sanity0/test.log' failed
make: Leaving directory '/media/D/NVDLA_20180131/hw-nvdlav1_v1/verif/sim'
logfile: outdir/build.log
==============================================
==================BUILD PASS==================
==============================================

So, what is the CMD-015 error???

CrazyBingo, 2018-02-28:
@zdraw

CrazyBingo, 2018-02-28:
modify the linux stack limit, the command is thus, not upon:
* soft stack 10240
* hard stack 10240

jwise, 2018-03-03:
Yeah, this sounds like a system configuration / VCS configuration error; I'm not sure how much we can really assist from this end.  Did increasing the stack limits help with runtime, too?