nvdla/hw #103: Simulation fails using VCS 2017

elham-skh, 2018-02-27:
I'm following http://nvdla.org/integration_guide.html#test-format .
After configuring Makefile in /hw/verif/sim/, I tried "make sanity". 
Part of the error message is as follows:

/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsPrintf'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsDoStackUnwind'
_csrc0.so: undefined reference to `M_VCSgd_IQZKC'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `vfs_fgets'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedAllocateGroup'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckCallocFunc'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `vfs_get_next_virtual_file'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `vfs::vfsReleaseVirtualFileList(VFILE_LIST*)'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsGetHook'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsCallocFunc'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsucli.so: undefined reference to `storeMiscTfCb'
_csrc0.so: undefined reference to `P_VCSgd_cwZZU_0_0'
_csrc0.so: undefined reference to `M_VCSgd_pkb2T'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `vfs_set_readonly'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `vfs_fopen'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `vfs::fopen(char const*, char const*, char const*, vcsfs_t, unsigned int)'
_csrc0.so: undefined reference to `SimDaiDir'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `snpsCurrentGroup'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `ZsGetHdlVariableHandleById'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `ZsGetNextSimulationTimeIfc'
/home/utils/vcs/N-2017.12/linux64/lib/libvcsnew.so: undefined reference to `ZsFinalize'
collect2: error: ld returned 1 exit status
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory `/home//nvdla/hw/verif/sim/csrc'
Make exited with status 2

Is it due to incompatibility with VCS 2017? 
Any help is very much appreciated.

elham-skh, 2018-02-27:
Adding -LDFLAGS -Wl,--no-as-needed to vcs parameters solved the issue, as suggested here: 
https://github.com/freechipsproject/rocket-chip/issues/21

CrazyBingo, 2018-02-28:
we have created a wchat group for nvdla discussion, if you are interesting, please give me your wchat ID. TKS

CrazyBingo, 2018-02-28:
your *so is not ok.  you should set the LD_LIBVRARY_PATH. may be run the vcs dependant, and verif the vcs first with a simple *.v wile help you set the environment

CrazyBingo, 2018-03-01:
it seems that you should tmake -build verif_sim first.
but I have field as I saide #102

elham-skh, 2018-03-03:
I had done  tmake -build verif_sim and also had set LD_LIBRARY_PATH (and tested other .v files). Simulation kept failing until I added  -LDFLAGS -Wl,--no-as-needed to vcs parameters.

jwise, 2018-03-03:
Hmm, I wonder why that `-Wl,--no-as-needed` is necessary.  Which parameter did you add it to?  We can try to put something about that in the documentation.

libingjian123, 2018-09-25:
my wchat ID :13581807473 ,would you please add me
 to the wchat group@CrazyBingo

smartchen2k, 2018-10-05:
> we have created a wchat group for nvdla discussion, if you are interesting, please give me your wchat ID. TKS

my wchat ID: 52986446,  would you please add me to the groupchat @CrazyBingo

peter-d, 2019-03-26:
I have add similar linking issues, but on different references, when building the simulator using my own installed systemC.

When pointing to the VCS provided systemC in tree.make, this fixed the error for me:

`SYSTEMC := $VCS_HOME/include/systemc231p`

huayangshiboqi, 2019-05-13:
> we have created a wchat group for nvdla discussion, if you are interesting, please give me your wchat ID. TKS

Hi, could I join the wechat group? My wechat ID: jie0qishi
Thanks

wang18112020007, 2019-07-10:
> we have created a wchat group for nvdla discussion, if you are interesting, please give me your wchat ID. TKS

Hi, could I join the nvdla discussion group?  My whchat ID :1109925862

killerzula, 2019-09-10:
@CrazyBingo please add me to the WeChat group. My id is: wxid_w92I86mcgbp322
There is an I in the id not a 1

luoxinjie1146, 2019-09-10:
Hi @CrazyBingo , could you please add me to the WeChat group? My id is lxjie1146
The first 'l' is lowercase 'L'.
Thanks!

Hassan313, 2019-11-16:
Hi  @CrazyBingo . Can you add me to the WeChat group? My id is wxid_2yni6v76ehad12.

Thank you very much.

strongwind312, 2019-11-28:
Hi @CrazyBingo . Could you add me to the WeChat group?  My WeChat id is y1006719486.

Thanks！