nvdla/hw #104: AXI-compliant interconnects in NVDLA

elham-skh, 2018-02-27:
From DLA documentations, I understand that DBB is an AXI-compliant interconnect. From logs and code, I infer that it is exactly AXI interconnect. In what ways DBB (and in general DLA NoC) is different from AXI4?

jwise, 2018-03-03:
DBB configurability is a subset of AXI, and has not undergone any AXI compliance testing.  We anticipate that it should work with any AXI interconnect -- and have tested it thoroughly with ours -- but we shy away from calling it AXI because we haven't done any conformance testing.

closed 2018-03-03 by jwise

elham-skh, 2018-03-03:
Thank you jwise for your answer. You said DBB is a **subset** of AXI. Would you mind giving example of functionalities of AXI that DBB doesn't support?