nvdla/hw #105: There is no DUT(NV_nvdla) instances information in the dump file

balabalatt, 2018-02-28:
Hi,

I can't find any DUT(NV_nvdla) information in the dump file, the dump file vcdplus.vpd only contains the testbench instances info. How can I get the DUT information in vpd file?
Thanks.

jwise, 2018-03-03:
@peng-peter-li can you have a look at this?

samplefood, 2018-03-05:
Look into verif/synth_tb/tb_top.v.  This is the $vcdpluson function. Remove the filtered top.* modules to see all of the signals.

balabalatt, 2018-03-06:
Hi @jwise,

I am able to look at the DUT info right now. Can you tell me the toggle start and end cycles for each testcase? I need to specify the saif_start_trans and saif_end_trans in makefile for dumping saif. e.g. the toggle start and end for diag conv_8x8_fc_int16 and others. Thanks!