nvdla/hw #109: NV_NVDLA_CMAC_CORE_MAC_mul doubt

IgnacioGoldman, 2018-03-20:
I would like to know how this module exactly works.

Supposing op_a_dat = 16h57e5
                  op_b_dat = 16h4242

Which are the values of res_a and res_b and why? 

(the simulation throws me: 
res_a = 32h6b6e4dca
res_b = 32h00000040)

pumpkinpbk, 2018-04-19:
in booth encode the MSB has been inverted, to get the final correct result you have to do the sign compensation, which CMAC is done in the level4 adder tree.

hope this would answer your question.

IgnacioGoldman, 2018-04-19:
First of all, thanks for answering. Unfortunately still not understanding.

I have moved into INT16 standard testbench conv_8x8 without any modifications and i have the following results

At booth mul level --> multiplication is straightforward
At MUL level --> im not having the expected results. For example:
inputs:     0xAAC3 and 0XE4CE (expected multiplication result 989F 14EA)
outputs:  0x5D18D4EA and 0x014A4000 (the sum is 5E63 14EA)

As you can see they are different and as you say, there's another CSA with the rest of the MUL units...
At MAC level --> given the image and the weight slice, after the multiply and accumulate process i'm expecting..               9 B098 63B1
but i'm having...     3F EC1D 63B1 
value that a proceed until the calculator of partition_a

Are this values ok? if they are ok, which is the relation between the expected value of a simple multiplication?

closed 2018-04-24 by IgnacioGoldman

ecqin, 2018-06-22:
Any information on this?

IgnacioGoldman, 2018-06-28:
NVDLA new builds of MAC cell units do not generate BOOTH multipliers explicitly, they are just dot (*) multiplicators. In other words, try to generate the last NVDLA vmod and check how he MAC is described.
(large or small version, is the same)