nvdla/hw #112: Verilator run have error happen help !!!!!!

pochen125, 2018-04-10:
I gave been running stimulation by Verilator. The error like this

%Warning-CASEINCOMPLETE: ../../outdir/nv_full/vmod/nvdla/csc/NV_NVDLA_CSC_WL_dec.v:12633: Case values incompletely covered (example pattern 0x0)
%Warning-CASEINCOMPLETE: ../../outdir/nv_full/vmod/nvdla/csc/NV_NVDLA_CSC_WL_dec.v:12766: Case values incompletely covered (example pattern 0x0)
%Warning-CASEINCOMPLETE: ../../outdir/nv_full/vmod/nvdla/csc/NV_NVDLA_CSC_WL_dec.v:12900: Case values incompletely covered (example pattern 0x0)
%Warning-CASEINCOMPLETE: ../../outdir/nv_full/vmod/nvdla/csc/NV_NVDLA_CSC_WL_dec.v:13035: Case values incompletely covered (example pattern 0x0)
%Error: Verilator threw signal 9.  Consider trying --debug --gdbbt
%Error: Command Failed /usr/local/bin/verilator_bin --cc --exe -f verilator.f --Mdir ../../outdir/nv_full/verilator/ nvdla.cpp --compiler clang --output-split 25000
Makefile:22: recipe for target '../../outdir/nv_full/verilator/VNV_nvdla.mk' failed
make: *** [../../outdir/nv_full/verilator/VNV_nvdla.mk] Error 255

jwise, 2018-04-10:
`%Error: Verilator threw signal 9. Consider trying --debug --gdbbt` means that it received a SIGKILL.  Probably your machine ran out of memory, or it hit a ulimit... make sure you are running in an environment that has at least 20GB of RAM, and that you have ulimits set to unlimited.

pochen125, 2018-04-12:
Could you tell me why %Warning-CASEINCOMPLETE: ../../outdir/nv_full/vmod/nvdla/csc/NV_NVDLA_CSC_WL_dec.v:13035: Case values incompletely covered (example pattern 0x0) happen at verilator running

jwise, 2018-04-12:
`%Warning-CASEINCOMPLETE` is a benign lint warning.