nvdla/hw #116: does NVDLA support reshape-transpose feature?

pumpkinpbk, 2018-04-19:


pumpkinpbk, 2018-04-19:
below is in the NVDLA primer, and transpose function is mentioned there:

"Data Reshape Engine 
The data reshape engine performs data format transformations (e.g., splitting or slicing, merging, contraction, reshape-transpose). Data in memory often needs to be reconfigured or reshaped in the process of performing inferencing on a convolutional network. For example, “slice” operations may be used to separate out different features or spatial regions of an image, and “reshape-transpose” operations (common in deconvolutional networks) create output data with larger dimensions than the input dataset.
The data reshape engine maps onto TensorFlow operations such as tf.nn.conv2d_transpose, tf.concat, tf.slice, and tf.transpose."

quasar-s, 2018-04-20:
Rubik is a dedicated unit for reshape-transpose, etc. Where in inference do you need reshape-transpose?

pumpkinpbk, 2018-04-22:
thank you guys for clarifying the problem. i'm not familiar with the inference flow, and were just learning the hardware part according to the primer.

closed 2018-04-22 by pumpkinpbk