nvdla/hw #120: Nv_small problem to build test bench

Y159753, 2018-04-23:
Hello, after following the build tree instructions for the Nv_small configuration I try to continue with the testbench build, but it doesn't work. Do you have an idea of the cause ? 

yago@Linux-2:~/hw_nv_small/hw$ ./tools/bin/tmake -build ready_for_test
[TMAKE]: building nv_small in spec/defs
[TMAKE]: building nv_small in spec/manual
[TMAKE]: building nv_small in verif/testbench/trace_generator
Traceback (most recent call last):
  File "../../../tools/bin/epython", line 162, in <module>
    process_file.exe_input()
  File "../../../tools/bin/epython", line 117, in exe_input
    self.codes_gen(tmp_file_hdl, scripts)
  File "../../../tools/bin/epython", line 146, in codes_gen
    codes = self.exe_python(scripts)
  File "../../../tools/bin/epython", line 154, in exe_python
    exec(scripts)
  File "<string>", line 4, in <module>
  File "../../.././verif/tools/spec2constrain.py", line 103, in enum_gen
    print('%-50s = \'h%0x' % (str0,keys[idx]))
TypeError: unicode argument expected, got 'str'
make: *** [../../../outdir/nv_small/verif/testbench/trace_generator/resources/nvdla_sdp_rdma_resource.sv] Error 1
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

yago@Linux-2:~/hw_nv_small/hw$ ./tools/bin/tmake -build verif_protection
[TMAKE]: building nv_small in spec/defs
[TMAKE]: building nv_small in spec/manual
[TMAKE]: building nv_small in verif/testbench/trace_generator
Traceback (most recent call last):
  File "../../../tools/bin/epython", line 162, in <module>
    process_file.exe_input()
  File "../../../tools/bin/epython", line 117, in exe_input
    self.codes_gen(tmp_file_hdl, scripts)
  File "../../../tools/bin/epython", line 146, in codes_gen
    codes = self.exe_python(scripts)
  File "../../../tools/bin/epython", line 154, in exe_python
    exec(scripts)
  File "<string>", line 4, in <module>
  File "../../.././verif/tools/spec2constrain.py", line 103, in enum_gen
    print('%-50s = \'h%0x' % (str0,keys[idx]))
TypeError: unicode argument expected, got 'str'
make: *** [../../../outdir/nv_small/verif/testbench/trace_generator/resources/nvdla_cc_dp_resource.sv] Error 1
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

leiwang2016, 2018-04-23:
@Y159753 are you using python 2 or python 3. You should be able to get rid of this error with python 3.

leiwang2016, 2018-04-24:
@HKLee2040 
python 2 string defaults to 8bits, which is fixed in python 3.
if you look at environment setup guide for v2. 
http://nvdla.org/hw/v2/environment_setup_guide.html
the tree.make template has python 3.
Enter python path (Press ENTER if use:
/home/tools/continuum/Anaconda3-5.0.1/bin/python):
So, I guess python 3 should work for v2.

Y159753, 2018-04-24:
You are both right. I linked the tree make with the wrong version of python. Thank you for such reactivity !

MINZHIJI, 2018-04-25:
I run `./tools/bin/tmake -build ready_for_test`, I get some error messages.
```
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in verif/testbench/trace_generator 
make: *** [vcs_build] Error 1
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE
```
And I trace the output log `outdir/nv_small.log`, I get the messages.
```
	However, due to incremental compilation, no re-compilation is necessary.
make[1]: Entering directory `/home/caslab/workspace/NVDLA/hw_4_25/hw/outdir/nv_small/verif/testbench/trace_generatorcsrc'
make[1]: Leaving directory `/home/caslab/workspace/NVDLA/hw_4_25/hw/outdir/nv_small/verif/testbench/trace_generatorcsrc'
make[1]: Entering directory `/home/caslab/workspace/NVDLA/hw_4_25/hw/outdir/nv_small/verif/testbench/trace_generatorcsrc'
rm -f _csrc*.so pre_vcsobj_*.so share_vcsobj_*.so
ld -shared  -o .//../trace_generator/simv.daidir//_csrc0.so objs/amcQw_d.o 
rm -f _csrc0.so
if [ -x /home/caslab/workspace/NVDLA/hw_4_25/hw/verif/testbench/trace_generator/../../../outdir/nv_small/verif/testbench/trace_generator/simv ]; then chmod -x /home/caslab/workspace/NVDLA/hw_4_25/hw/verif/testbench/trace_generator/../../../outdir/nv_small/verif/testbench/trace_generator/simv; fi
g++  -o /home/caslab/workspace/NVDLA/hw_4_25/hw/verif/testbench/trace_generator/../../../outdir/nv_small/verif/testbench/trace_generator/simv   -Wl,-rpath-link=./ -Wl,-rpath='$ORIGIN'/simv.daidir/ -Wl,-rpath=./simv.daidir/ -Wl,-rpath='$ORIGIN'/simv.daidir//scsim.db.dir  -rdynamic  /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/vpdlogstub.o fp_func_dpi.o uvm_dpi.o     _29755_archive_1.so _prev_archive_1.so _csrc0.so  SIM_l.o  _csrc0.so    rmapats_mop.o rmapats.o rmar.o rmar_nd.o  rmar_llvm_0_1.o rmar_llvm_0_0.o          /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libzerosoft_rt_stubs.so /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libvirsim.so /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/liberrorinf.so /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libsnpsmalloc.so /home/caslab/workspace/tools/CIC_tools/verdi/cur/share/PLI/VCS/LINUX64/pli.a    /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libvcsnew.so /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libsimprofile.so /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libuclinative.so /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/vcs_tls.o   -Wl,-whole-archive /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/libvcsucli.so -Wl,-no-whole-archive       ./../trace_generator/simv.daidir/vc_hdrs.o  _vcs_pli_stub_.o   /home/caslab/workspace/tools/CIC_tools/vcs/cur/linux64/lib/vcs_save_restore_new.o -ldl -lm  -lc -lpthread -ldl 
_prev_archive_1.so: undefined reference to `_C_f72f649e_0'
_prev_archive_1.so: undefined reference to `_C_794bcdca_0'
_prev_archive_1.so: undefined reference to `_C_420375c4_0'
...
``` 
What is wrong on my steps? Thanks!

JunningWu, 2018-04-28:
@MINZHIJI Same error as yours. have you solved it?

ewq7899987, 2018-04-28:
Try to add the **-LDFLAGS -Wl,--no-ad-needed** parameters in the VCS_ARGS nvdla/hw/verif/testbench/trace_generator/Makefile  (at about line 122)

ewq7899987, 2018-04-28:
Try to add the -LDFLAGS -Wl,--no-ad-needed parameters in the VCS_ARGS nvdla/hw/verif/testbench/trace_generator/Makefile (at about line 122)

and the same to the trac_player

ewq7899987, 2018-04-28:
Try to add the **-LDFLAGS -Wl,--no-ad-needed** parameters in the VCS_ARGS nvdla/hw/verif/testbench/trace_generator/Makefile (at about line 122)

and the same to the trace_player

MINZHIJI, 2018-04-28:
@ewq7899987 
**--no-ad-needed**  is not found.
Is it **--no-as-needed**?

redpanda3, 2018-04-28:
My problem of nv_small is that no file in spec/def could be generated.

hongyongzhi, 2018-05-04:
-LDFLAGS -Wl,--no-as-needed has been added in VCS_ARGS，but there is nothing has been changed，why?

JunningWu, 2018-05-10:
@MINZHIJI Have you ever solved the error related to 
···
_prev_archive_1.so: undefined reference to
···

MINZHIJI, 2018-05-10:
@JunningWu 

I solved by **ewq7899987**'s method
> Try to add the `-LDFLAGS -Wl,--no-as-needed` parameters in the `VCS_ARGS` `nvdla/hw/verif/testbench/trace_generator/Makefile` (at about line 122)
and the same to the `trace_player`

kouzhentao, 2018-05-31:
@MINZHIJI @ewq7899987 can u please paste the line how to add this -LDFLAGS -Wl,--no-as-needed

Is this WI or W1 or Wl? does this "," needed? thanks

MINZHIJI, 2018-05-31:
@kouzhentao 
`path/to/nvdla/hw/verif/testbench/trace_generator/Makefile`
[image]

quasar-s, 2018-05-31:
@MINZHIJI , Thanks! For me it was designware DW02_multp,v. I disabled designware in tree.make, but I wish there was a better way to get around this.

koroyuki, 2018-07-18:
Dear Sirs,
I followed @kouzhentao way. An uvm related error, however, remains.
I appreciate if I have a suggestion.
Thanks in advance.

[1] test program compilation 
$ ./tools/bin/tmake -build ready_for_test
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in verif/testbench/trace_generator 
make: *** [vcs_build] Error 255
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE
$

[2] Error in  ../outdir/nv_small.log
$ cat nv_small.log
/autohome/ozuser/AP_lib/python/python-3.5.2/bin/python3.5 ../../tools/bin/defgen -i /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.def -o /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.h -b c
/autohome/ozuser/AP_lib/python/python-3.5.2/bin/python3.5 ../../tools/bin/defgen -i /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.def -o /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.vh -b v
/autohome/ozuser/AP_lib/python/python-3.5.2/bin/python3.5 ../../tools/bin/defgen -i /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.def -o /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.py -b py
/autohome/ozuser/AP_lib/python/python-3.5.2/bin/python3.5 ../../tools/bin/defgen -i /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.def -o /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.mk -b mk
/autohome/ozuser/AP_lib/python/python-3.5.2/bin/python3.5 ../../tools/bin/defgen -i /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.def -o /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs/project.pm -b pm
==============================================
files are generated under /autohome/ozuser/APD2/tago/nvdla_hw_master-apl/outdir/nv_small/spec/defs
==============================================

---- omitted ----

Mdir=../../../outdir/nv_small/verif/testbench/trace_generatorcsrc
*** Using c compiler /bin/g++ instead of cc ...
                         Chronologic VCS (TM)
       Version G-2012.09-SP1_Full64 -- Wed Jul 18 15:44:56 2018
               Copyright (c) 1991-2012 by Synopsys Inc.
                         ALL RIGHTS RESERVED

This program is proprietary and confidential information of Synopsys Inc.
and may be used and disclosed only as authorized in a license agreement
controlling such use and disclosure.

Warning-[UVM_WVER] Wrong UVM version
  Option 'uvm-1.2' is not recognized. Supported UVM versions are 'uvm-1.0' and
  'uvm-1.1'.

Warning-[ACC_CLI_ON] ACC/CLI capabilities enabled
  ACC/CLI capabilities have been enabled for the entire design. For faster 
  performance enable module specific capability in pli.tab file

Error-[SFCOR] Source file cannot be opened
  Source file "/uvm_pkg.sv" cannot be opened for reading due to 'No such file 
  or directory'.
  Please fix above issue and compile again.

2 warnings
1 error
CPU time: .050 seconds to compile
make: *** [vcs_build] Error 255
$ 

---- End ---

zackgd1024, 2018-08-24:
After adding ”-LDFLAGS -Wl,--no-as-needed“， I still get the following error:
_prev_archive_1.so: undefined reference to `_C_cda4aa8b_0'
_prev_archive_1.so: undefined reference to `_C_21b0f3c2_0'
collect2: error: ld returned 1 exit status
Makefile:104: recipe for target 'product_timestamp' failed

Does anyone also meet with the issue?
Thanks

smartchen2k, 2018-10-08:
> After adding ”-LDFLAGS -Wl,--no-as-needed“， I still get the following error:
> _prev_archive_1.so: undefined reference to `_C_cda4aa8b_0' _prev_archive_1.so: undefined reference to`_C_21b0f3c2_0' [...]

hi, did you solve that??

peter-d, 2019-03-26:
I got this error when building the simulator using my own installed systemC.
When pointing to the VCS provided systemC in tree.make, this fixed the error for me:

`SYSTEMC := $VCS_HOME/include/systemc231p`

koroyuki, 2019-03-29:
Dear Sir,
I see.
By the way what OS, shell, and system-c version  do you use?

MengXiao92, 2019-03-31:
Dear Sir,
I got a problem, when I build the nv_small_256, by modified the tree.make 
PROJECTS := nv_small_256

run the command   ./tools/bin/tmake -build ready_for_test

got the error 
make: *** [vcs_build] Error 255
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE
 
the log in the file nv_small_256.verif_trace_player.log
$ cat nv_small_256.verif_trace_player.log

Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/NV_DW02_tree.v'
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/NV_DW_lsd.v'
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/NV_DW_minmax.v'
Parsing library directory file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/rams/model/RAMDP_128X11_GL_M2_E2.v'
Parsing library directory file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/rams/model/RAMDP_80X14_GL_M2_E2.v'
Parsing library directory file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/rams/model/RAMPDP_160X16_GL_M2_D2.v'
Parsing library directory file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/rams/model/RAMPDP_160X65_GL_M2_D2.v'
Parsing library directory file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/rams/synth/nv_ram_rwsp_80x65_logic.v'
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v'
Parsing included file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/include/simulate_x_tick.vh'.
Back to file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v'.
Parsing included file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/include/simulate_x_tick.vh'.
Back to file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v'.
Parsing included file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/include/simulate_x_tick.vh'.
Back to file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v'.
Parsing included file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/include/simulate_x_tick.vh'.
Back to file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/nvdla/nocif/NV_NVDLA_XXIF_libs.v'.
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/RANDFUNC.vlib'
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/nv_assert_no_x.vlib'
Parsing included file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/assertion_header.vh'.
Back to file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/nv_assert_no_x.vlib'.
Parsing included file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/assertion_task.vh'.
Back to file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/nv_assert_no_x.vlib'.
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/NV_DW02_tree.v'
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/NV_DW_lsd.v'
Parsing library file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/vlibs/NV_DW_minmax.v'
Parsing library directory file '/home/mengxiao/NVDLA/hw-nv_small1/outdir/nv_small_256/vmod/rams/model/RAMPDP_80X66_GL_M1_D2.v'
1 error
CPU time: 1.774 seconds to compile
Makefile:330: recipe for target 'vcs_build' failed
make: *** [vcs_build] Error 255

And the nv_small is OK, I want to know what wrong with nv_small_256 
Does anyone also meet with the issue?
Thanks

peter-d, 2019-04-01:
@MengXiao92 : probably best to check `outdir/nv_small_256.log` to get a better view on the error. It might have more info.
@koroyuki : Not sure how much this matters, but I am running RHEL:
```
$ less /etc/redhat-release
Red Hat Enterprise Linux Server release 7.3 (Maipo)
$ uname -a
Linux inter006.imec.be 3.10.0-514.16.1.el7.x86_64 #1 SMP Fri Mar 10 13:12:32 EST 2017 x86_64 x86_64 x86_64 GNU/Linux
```
SystemC is v2.3.1 provided by VCS (v2017.03-1)

jonesduan, 2019-06-23:
> hi, did you solve that??

rm -rf outdir
and do it all again