nvdla/hw #121: Build testbench for nvdla_master and miss variable declaration

kcchang76, 2018-04-25:
Hello, I've tried to build testbench and found error.
The error log (nv_small.verif_trace_generator.log) shows that the identifier 'kernel_stride_height' in nvdla_pdp_resource.sv is not declared.
By checking nvdla_pdp_resource.sv in outdir, it seems the declaration of 'kernel_stride_height' disappers.
Do you have any idea about this error?

Linux:kcchang/nvdla_master/hw>./tools/bin/tmake -build ready_for_test
[TMAKE]: building nv_small in spec/defs
[TMAKE]: building nv_small in spec/manual
[TMAKE]: building nv_small in verif/testbench/trace_generator
make: Warning: File `/home/kcchang/nvdla_master/hw/outdir/nv_small/spec/defs/project.mk' has modification time 1.3e+03 s in the future
make: *** [vcs_build] Error 255
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

*** content of outdir/nv_small.verif_trace_generator.log ***
Error-[IND] Identifier not declared
/home/kcchang/nvdla_master/hw/outdir/nv_small/verif/testbench/trace_generator/resources/nvdla_pdp_resource.sv, 284
  Identifier 'kernel_stride_height' has not been declared yet. If this error
  is not expected, please check if you have set `default_nettype to none.

*** content of outdir/nv_small/verif/testbench/trace_generator/resources/nvdla_pdp_resource.sv ***
Line #82~86: 
    rand kernel_width_t             kernel_width;
    rand kernel_height_t            kernel_height;
    rand bit [3:0]                  kernel_stride_width;
    rand bit [16:0]                 recip_kernel_width;
    rand bit [16:0]                 recip_kernel_height;

*** content of verif/testbench/trace_generator/resources/nvdla_pdp_resource.sv ***
Line #82~87: 
    rand kernel_width_t             kernel_width;
    rand kernel_height_t            kernel_height;
    rand bit [3:0]                  kernel_stride_width;
    rand bit [3:0]                  kernel_stride_height;
    rand bit [16:0]                 recip_kernel_width;
    rand bit [16:0]                 recip_kernel_height;

quasar-s, 2018-05-31:
I get the exact same error: 
[TMAKE]: building nv_small in verif/vip/reference_model/nvdla_cmod_wrap 
[TMAKE]: building nv_small in verif/testbench/trace_player 
make: *** [vcs_build] Error 255
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

Did you figure out what was the cause?

Thanks!

kcchang76, 2018-06-01:
Update my finding for someone who gets the same error.

The error occur because the following generated file is incomplete:
  outdir/nv_small/spec/manual/opendla.vh
=> the following 2 definitions are not generated
NVDLA_PDP_D_POOLING_KERNEL_CFG_0_KERNEL_STRIDE_HEIGHT_RANGE 23:20
NVDLA_PDP_D_POOLING_KERNEL_CFG_0_KERNEL_STRIDE_HEIGHT_SIZE 4

The opendla.vh file is generated by /spec/manual/ORDT_xml2others.pl,
which is called In line 94 of /spec/manual/RDL_merge.pl.

I'm not familiar with perl, but when I run this in the other Linux system, it passes.

kcchang76, 2018-06-01:
Hi,

I’ve update my finding in github.
FYI.

K.C.

本信件可能包含工研院機密資訊，非指定之收件者，請勿使用或揭露本信件內容，並請銷毀此信件。 This email may contain confidential information. Please do not use or disclose it in any way and delete it if you are not the intended recipient.

quasar-s, 2018-06-05:
Here's another set of errors with nv_large:
Identifier 'NVDLA_MCIF_BURST_SIZE_LOG2' has not been declared yet.
Identifier 'NVDLA_MCIF_BURST_SIZE' has not been declared yet. 

Are we supposed to manually define them in spec???

cqlq2020, 2020-07-05:
I saw this issue too when I tried to build nv small(v2). Finally I found it's related to python version. When I used python version 3.8.1, I saw this exactly "Error-[IND] Identifier not declared
/home/kcchang/nvdla_master/hw/outdir/nv_small/verif/testbench/trace_generator/resources/nvdla_pdp_resource.sv, 284
Identifier 'kernel_stride_height' has not been declared yet"

Later I changed python version to 3.4, then this issue is gone. 

btw, for v1, nv_full, I used python version 2.7, it's fine.