nvdla/hw #122: I can not synthesis...

sounansu, 2018-04-25:
I try to compile at nv_small_256 project, using

./syn/scripts/syn_launch.sh -config syn/templates/config.sh -mode wlm -project nv_small_256

command.

But, I can not compile 'c' partition.

At log file.
nvdla_syn_20180425_1647/log/NV_NVDLA_partition_c_wlm.log
-------------------------------------------------------------------------
Error:  nvdla_syn_20180425_1647/src/NV_NVDLA_CDMA_wt.v:1592: case equality (===) is not supported by synthesis. (VER-189)
Error:  nvdla_syn_20180425_1647/src/NV_NVDLA_CDMA_wt.v:1606: case equality (===) is not supported by synthesis. (VER-189)
Error:  nvdla_syn_20180425_1647/src/NV_NVDLA_CDMA_wt.v:1620: case equality (===) is not supported by synthesis. (VER-189)
Error:  nvdla_syn_20180425_1647/src/NV_NVDLA_CDMA_wt.v:1634: case equality (===) is not supported by synthesis. (VER-189)
Opening include file nvdla_syn_20180425_1647/src/simulate_x_tick.vh
Error:  Cannot recover from previous errors. (VER-518)
*** Presto compilation terminated with 5 errors. ***
--------------------------------------------------------------------------
There are '===' codes in NV_NVDLA_CDMA_wt.v

regards.

jwise, 2018-04-25:
Do you also get the issue attempting to synthesize `nv_small`?  `nv_small_256` is not ready for synthesis yet.

sounansu, 2018-04-26:
Thank you jwise!.

Now, I tried nv_small project.

There were no 'not supported by synthesis' erros.

But, 
---------------------------------------------------------
Error: No good flip-flop found in library. Cannot model
        the design. (RTDC-27)
---------------------------------------------------------
are occured at 'm' partition.

Now, I asked our EDA team about this error.

regards.

sounansu, 2018-05-08:
I asked our eda team.
So, my script lack of ff cell library.
And I add that library, it is ok to synthesis!.

Now, I can synthesis nv_small and nv_full.
But, I can not synthesis nv_large......

I will tray to modify nv_small configuration.

Thank you all!

nodushiv, 2018-05-13:
Closing the issue. Thanks!

closed 2018-05-13 by nodushiv