nvdla/hw #124: Unit conflict found about cap unit

JunningWu, 2018-04-28:
when compile using dc, I got the Error about Constraint cap unit and main library's, which is 1.000000e-04pf VS pF.

I wonder where to set the cap unit, since I doesn't find it in the scripts.

nodushiv, 2018-05-03:
What is the error message you are seeing in DC?

JunningWu, 2018-05-04:
@nodushiv from the dc log file, the error message is this

```
ERROR: Unit conflict found: Constraint cap unit is '1.000000e-04pf'; main library cap unit is 'pF'. (IFS-001)
```
like what I have said, when changing the .sdc file, the cap unit in the log file is changed likewise.

nodushiv, 2018-05-13:
@JunningWu This usually happens when the target_library and link_library are not set correctly, and DC is unable to derive the units from the library. 

Can you ensure that the libraries are being loaded correctly?

If you can attach the whole log file, that would be ideal for debug.

nodushiv, 2018-05-26:
Closing this issue since I haven't heard back from @JunningWu 
Please reopen and provide more information if this remains to be an issue.

closed 2018-05-26 by nodushiv