nvdla/hw #127: nv_small build issue

thefpgaguy, 2018-05-10:
Hi, 
I'm trying to build nv_small and had the following error. Anyone encounter this before?

Perl version : This is perl, v5.10.1 (*) built for x86_64-linux-thread-multi

Krek

[TMAKE]: building nv_small in vmod/nvdla/cbuf 
[TMAKE]: building nv_small in vmod/nvdla/csc 
Processing NV_NVDLA_CSC_slcg.v into ../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_slcg.v.vcp
../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_slcg.v.vcp successfully written
eperl: ../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_slcg.v modified.
Processing NV_NVDLA_CSC_SG_dat_fifo.v into ../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_SG_dat_fifo.v.vcp
../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_SG_dat_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_SG_dat_fifo.v modified.
Processing NV_NVDLA_CSC_dl.v into ../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_dl.v.vcp
../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_dl.v.vcp successfully written
ERROR: eperl: eval:
&eperl::flop("-nodeclare -wid ${quat} -norst -en \"dat_rsp_sft_d2_en\" -d \"dat_rsp_l1_sft\" -q dat_rsp_l1_sft_d2");
&eperl::flop("-nodeclare -wid ${quat} -norst -en \"dat_rsp_sft_d3_en\" -d \"dat_rsp_l1_sft_d2\" -q dat_rsp_l1_sft_d3");

error:
Global symbol "$quat" requires explicit package name at (eval 39) line 1, <IN1> line 1777.
Global symbol "$quat" requires explicit package name at (eval 39) line 2, <IN1> line 1777.

 at ../../../tools/bin/eperl line 265
	eperl_main::Error('eval:\x{a}&eperl::flop("-nodeclare -wid ${quat} -norst -en \"dat_...') called at ../../../tools/bin/eperl line 210
	eperl_main::EvalScript('&eperl::flop("-nodeclare -wid ${quat} -norst -en \"dat_rsp_sf...') called at ../../../tools/bin/eperl line 189
	eperl_main::ProcessInput('../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_dl.v.vcp', '../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_dl.v.vcp...') called at ../../../tools/bin/eperl line 158
make: *** [../../../outdir/nv_small/vmod/nvdla/csc/NV_NVDLA_CSC_dl.v] Error 25
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

eruanno123, 2018-05-10:
Looks like extra new line between perl scripts (NV_NVDLA_CSC_dl: line 1774 and 1777).

thefpgaguy, 2018-05-11:
Thanks @eruanno123 

I deleted 2 empty lines in hw/vmod/nvdla/csc/NV_NVDLA_CSC_dl.v
(lines 1774 and 1777) and the problem went away.

However I encountered another problem. This is a basic build fresh from Git, could it be versions of my perl? or plugins?

Perl : 5.10.1
Capture::Tiny "VERSION: 0.48"
 XML::Simple "VERSION: 2.25"

   
The new problem:

Processing NV_NVDLA_CDP_DP_bufferin_tp1.v into ../../../outdir/nv_small/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_bufferin_tp1.v.vcp
../../../outdir/nv_small/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_bufferin_tp1.v.vcp successfully written
ERROR: eperl: eval:
foreach my $k (0..4) {
my $m = $k + 4;
foreach my $j (0..7) {
print qq(
data_shift_${m}${j} <= {${tp}*${bpe}{1'd0}};
);
}
}

error:
Global symbol "$tp" requires explicit package name at (eval 12) line 4, <IN1> line 347.
Global symbol "$bpe" requires explicit package name at (eval 12) line 4, <IN1> line 347.

 at ../../../tools/bin/eperl line 265
	eperl_main::Error('eval:\x{a}foreach my $k (0..4) {\x{a}my $m = $k + 4;\x{a}foreach my $j (0...') called at ../../../tools/bin/eperl line 210
	eperl_main::EvalScript('foreach my $k (0..4) {\x{a}my $m = $k + 4;\x{a}foreach my $j (0..7) {...') called at ../../../tools/bin/eperl line 189
	eperl_main::ProcessInput('../../../outdir/nv_small/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_buffe...', '../../../outdir/nv_small/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_buffe...') called at ../../../tools/bin/eperl line 158
make: *** [../../../outdir/nv_small/vmod/nvdla/cdp/NV_NVDLA_CDP_DP_bufferin_tp1.v] Error 25
Died at tools/bin/tmake line 254.

mmaciag, 2018-05-11:
I looked into *.v.vcp files and those new lines seem to be merged. Check GCC version, maybe the preprocessor (cpp) behaves differently. The recommended version is gcc-4.9 AFAIK.

thefpgaguy, 2018-05-11:
Thanks @mmaciag 

I was using gcc version 4.4.7 and when I switch to gcc version 6.3.1 I was able to build it.
I keep thinking its a perl issue.

Thanks for the big help.

Krek

closed 2018-05-11 by thefpgaguy

owenwu99, 2018-07-24:
5 cents experience sharing: I encountered the same issue described by thefpgaguy. At first, it looks like a Perl issue since the error message from Perl said so. But after I switch the cpp version from 4.4.7 to 4.7.2, this issue went away and I were able to build it.