nvdla/hw #128: Synthesis: Any guidance on setting up SGE?

silvaurus, 2018-05-11:
Hi! I'm trying to synthesize NVDLA using command syn/scripts/syn_launch.sh -mode dct -config syn/templates/config.sh, but I keep getting errors on SGE such as "qsub: command not found", "Please set the environment variable SGE_ROOT", "cell directory "/usr/share/gridengine/default" doesn't exist"... I have not used SGE before, so I'm wondering if there could be any guidance on installing and setting up SGE? Thank you so much!

nodushiv, 2018-05-13:
Please see the documentation for the `COMMAND_PREFIX` variable in the Integrator's manual:
http://nvdla.org/hw/v1/integration_guide.html#synthesis

You can use that variable to dispatch the job to whatever farm you want.
The `COMMAND_PREFIX` variable is prefixed to the dc_shell command. 

If you want, you can change that variable to an empty string, and all jobs will run on the local machine, sequentially.

silvaurus, 2018-05-17:
Thank you so much! I just found I only changed the variable in default_config.sh, but didn't change it in templates/config.sh.

nodushiv, 2018-05-26:
No problem. Closing this issue.

closed 2018-05-26 by nodushiv