nvdla/hw #131: tools/bin/tmake -build cmod_top - can't build cmod_top

darrenleong, 2018-05-15:
Hi, 

when i try to build cmod_top, i faced the following error:

../cmod/hls/include/ac_fixed.h:410:108: error invalid operands of types '<unresolved overloaded function type>' and 'int' to binary 'operator<'

someone encountered the same problem? i am using gcc version 7.2.0, or i must use 4.9.3 as NVDLA opensource suggested?

cathypa4u, 2018-06-05:
I have the same error in building cmod_top.

Followings are build logs.

========================================================================
make: 디렉터리 '/media/swahn4u/Dev/TOOL/NVDLA/hw/spec/defs' 들어감
/usr/bin/gcc -undef -nostdinc -P -C nv_full.spec -o /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/defs/project.def
Makefile:22: recipe for target '/media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/defs/project.def' failed
make: 디렉터리 '/media/swahn4u/Dev/TOOL/NVDLA/hw/spec/defs' 나감
make: 디렉터리 '/media/swahn4u/Dev/TOOL/NVDLA/hw/spec/manual' 들어감
/usr/bin/java -jar Ordt.jar -parms test.parms -systemverilog /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/sv/ -verilog /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_v.v -uvmregs /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_ral.sv -cppmod /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod -cppdrvmod /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod test.rdl
Open Register Design Tool, version=170915.01, input=test.rdl
Ordt: reading parameters from test.parms...
Ordt: building verilog...
Ordt: writing verilog file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_v.v...
Ordt: building systemverilog...
Ordt: writing systemverilog file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/sv/simple1_pio.sv...
Ordt: writing systemverilog file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/sv/simple1_jrdl_logic.sv...
Ordt: writing systemverilog file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/sv/simple1_jrdl_decode.sv...
Ordt: building UVM regs...
Ordt: writing UVM regs file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_ral.sv...
Ordt: building C++ model...
Ordt: writing C++ model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod/ordt_pio_common.hpp...
Ordt: writing C++ model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod/ordt_pio_common.cpp...
Ordt: writing C++ model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod/ordt_pio.hpp...
Ordt: writing C++ model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod/ordt_pio.cpp...
Ordt: building C++ driver model...
*** INFO ***: Overlay 0 total processed instances=16, unique instances=16, duplicate instances=0
Ordt: writing C++ driver model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod/ordt_pio_common.hpp...
Ordt: writing C++ driver model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod/ordt_pio_common.cpp...
Ordt: writing C++ driver model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod/ordt_pio_drv.hpp...
Ordt: writing C++ driver model file /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod/ordt_pio_drv.cpp...
Ordt complete Tue Jun 05 19:45:07 KST 2018
#@cp -f /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_v.v /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/; rm /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_v.v -rf
#@cp -f /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_ral.sv /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/; rm /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/regs_ral.sv -rf
#@cp -rf /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/sv/ /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/; rm /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/sv/ -rf
#@cp -rf /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/; rm /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/cmod -rf
#@cp -rf /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/; rm /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual/dmod -rf
==============================================
files are generated under /media/swahn4u/Dev/TOOL/NVDLA/hw/outdir/nv_full/spec/manual
==============================================
make: 디렉터리 '/media/swahn4u/Dev/TOOL/NVDLA/hw/spec/manual' 나감
make: 디렉터리 '/media/swahn4u/Dev/TOOL/NVDLA/hw/cmod' 들어감
mkdir -p ../outdir/nv_full/cmod/../cmod/hls_wrapper/
/usr/bin/gcc -I./include -I../cmod/include/nvdla_ness_header -I../cmod/include -I../cmod/nvdla_payload -I../cmod/bdma -I../cmod/cacc -I../cmod/cbuf -I../cmod/cdma -I../cmod/cdp -I../cmod/cmac -I../cmod/csb_master -I../cmod/csc -I../cmod/cvif -I../cmod/glb -I../cmod/hls_wrapper -I../cmod/mcif -I../cmod/nvdla_core -I../cmod/pdp -I../cmod/rubik -I../cmod/sdp -I../cmod/bdma/gen -I../cmod/cacc/gen -I../cmod/cbuf/gen -I../cmod/cdma/gen -I../cmod/cdp/gen -I../cmod/cmac/gen -I../cmod/csb_master/gen -I../cmod/csc/gen -I../cmod/cvif/gen -I../cmod/glb/gen -I../cmod/hls_wrapper/gen -I../cmod/mcif/gen -I../cmod/nvdla_core/gen -I../cmod/pdp/gen -I../cmod/rubik/gen -I../cmod/sdp/gen -I../cmod/hls/include -I../cmod/hls/cdma_libs -I../cmod/hls/cdp_libs -I../cmod/hls/csc_libs -I../cmod/hls/sdp -I../cmod/hls/vlibs -I/usr/local/systemc-2.3.0//include -MMD -MP -fPIC -Wall -Werror -DSC_INCLUDE_DYNAMIC_PROCESSES -Wp,-w -std=c++11  -c ../cmod/hls_wrapper/cdma_hls_wrapper.cpp -o ../outdir/nv_full/cmod/../cmod/hls_wrapper/cdma_hls_wrapper.cpp.o
Makefile:165: recipe for target '../outdir/nv_full/cmod/../cmod/hls_wrapper/cdma_hls_wrapper.cpp.o' failed
make: 디렉터리 '/media/swahn4u/Dev/TOOL/NVDLA/hw/cmod' 나감
========================================================================

bullit2, 2018-06-25:
I have the same issue -- any idea ?

darrenleong, 2018-06-25:
i can now build the cmod_top after changed my VM to ubuntu16.04.4 (using its default cpp/gcc/g++)  and manually installed java jdk1.8.

bullit2, 2018-06-25:
I am running on 18.0.4 and it fails : was that the original ubuntu version you had when failing initially ?

jhwangus, 2018-08-29:
I also have the same issue on Ubuntu 18.04 and g++/gcc/cpp 7.3.0, Jdk 1.8.0_181.

I think cmod/hls/include/ac_fixed.h need the following change:

--- ac_fixed_orig.h	2018-08-29 14:47:59.977317789 -0700
+++ ac_fixed.h	2018-08-29 14:35:51.514589435 -0700--- ac_fixed_orig.h	2018-08-29 14:47:59.977317789 -0700
+++ ac_fixed.h	2018-08-29 14:35:51.514589435 -0700
@@ -407,7 +407,7 @@
 #endif
 
   // Explicit conversion functions to ac_int that captures all integer bits (bits are truncated)
-  inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).slc<AC_MAX(I,1)>(0); }
+  inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).template slc<AC_MAX(I,1)>(0); }
 
   // Explicit conversion functions to C built-in types -------------
   inline int to_int() const { return ((I-W) >= 32) ? 0 : (signed int) to_ac_int(); } 

@@ -407,7 +407,7 @@
 #endif--- ac_fixed_orig.h	2018-08-29 14:47:59.977317789 -0700
+++ ac_fixed.h	2018-08-29 14:35:51.514589435 -0700
@@ -407,7 +407,7 @@
 #endif
 
   // Explicit conversion functions to ac_int that captures all integer bits (bits are truncated)
-  inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).slc<AC_MAX(I,1)>(0); }
+  inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).template slc<AC_MAX(I,1)>(0); }
 
   // Explicit conversion functions to C built-in types -------------
   inline int to_int() const { return ((I-W) >= 32) ? 0 : (signed int) to_ac_int(); } 

 
   // Explicit conversion functions to ac_int that captures all integer bits (bits are truncated)
-  inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).slc<AC_MAX(I,1)>(0); }
+  inline ac_int<AC_MAX(I,1),S> to_ac_int() const { return ((ac_fixed<AC_MAX(I,1),AC_MAX(I,1),S>) *this).template slc<AC_MAX(I,1)>(0); }
 
   // Explicit conversion functions to C built-in types -------------
   inline int to_int() const { return ((I-W) >= 32) ? 0 : (signed int) to_ac_int(); }

JunningWu, 2018-08-30:
every thing is ok for me, ubuntu 14.04, vmware
```
System Requirements
It is recommended to build the virtual simulator in a system that meets the following requirements:
OS: Ubuntu 14.04
g++ 4.8.4
git > 1.8.2
systemc 2.3.0
cmake > 2.8
libboost > 1.34
python dev
glib2 dev
pixman dev
lua 5.2 dev
swig
libcap dev
libattr1 dev
```

mmaciag, 2018-09-03:
Hi, just to cross-reference: https://github.com/nvdla/hw/pull/191#issuecomment-411647611. Newer ac_types library seems to fix language issues in newer compilers.

abhishek42, 2019-06-26:
@mmaciag This was helpful. I updated the ac_types in /cmod/hls/include and the compilation went through

killerzula, 2019-09-09:
@jhwangus  Thanks a lot for your reply, saved me a lot of headache.

prasshantg, 2019-09-10:
@darrenleong is this issue resolved? can it be closed?