nvdla/hw #133: Neural Net integrations open questions

Y159753, 2018-05-16:
You don't discuss so much about that in the documentation, I understand it is not a closed topic but I would like to have some hints about this :

How to optimize the hardware for a given neural net ? What kind of approach would you advise ?

Does any neural net will fit on any configuration of the NVDLA but at the cost of big timing delay or power consumption ?

jwise, 2018-05-16:
As you noted, the area is quite broad indeed.  But we do have a performance estimator for various configurations of NVDLA, which can help you understand the various tradeoffs of different NVDLA configurations: https://github.com/nvdla/hw/blob/nvdlav1/perf/NVDLA_OpenSource_Performance.xlsx

In theory, any NVDLA can run any neural net that it supports the precision of (for instance, nv_small cannot run fp16 nets, because it does not have fp16 support); as you note, the tradeoff is mostly power consumption and performance.

Y159753, 2018-05-22:
Thank you for your answer, I will use this sheet as a starting point.

You say that the NVDLA can run any neural net with the the limit of  precision.
Is it too obvious for you that the NVDLA is dedicated to CNN because convolutions operations are optimised ?    
Or do you mean that it can also run in an optimised manner nets with RNN, LSTM or any of those http://www.asimovinstitute.org/neural-network-zoo/  ?