nvdla/hw #135: How to Include DesignWare IPs In Synthesis

silvaurus, 2018-05-17:
Hi! I'm wondering how to include DesignWare IPs in Synthesis. Currently I got an error like this:
Warning: Cannot find the design 'DW_lsd' in the library 'WORK'. (LBR-1)
I have DesignWare installed under Design Compiler, but I'm not sure if there are any global variables I should set up, or if I should include DW's source code folder in EXTRA_RTL.

Thank you so much!

nodushiv, 2018-05-26:
I assume that you aren't changing the `DESIGNWARE_NOEXIST RTL macro between build and synthesis. 

What have you set as LINK_LIB in the configuration file?

If you have a standard DC install, you should see these libraries:
${DC_PATH}/../libraries/syn/dw_foundation.sldb
${DC_PATH}/../libraries/syn/gtech.db
${DC_PATH}/../libraries/syn/standard.sldb

You should add them to LINK_LIB if you dont already. 

If that doesnt work - Can you post your DC log?

Thanks!

silvaurus, 2018-05-27:
Thank you so much! It works now!

nodushiv, 2018-06-05:
Thanks. Closing.

closed 2018-06-05 by nodushiv