nvdla/hw #136: Question On CMAC Input Shadowing

silvaurus, 2018-05-17:
Hi! I'm wondering if I can get some intuitive help from understanding CMAC input shadowing. Currently I know that it reads in 128 8 bit input data/weights, and output 8 1024 data/weight. And I understand how it generates data for specific data types. What I don't understand is the process of pre-selection. 

According to Unit Description, One NV_NVDLA_cmac is performing one atom operation of 8 kernels. For each MAC_cell it performs atom operation on one kernel. So those 8 MAC_cells should have the same input data while different input weights. Then I don't know if we just read in 128 x 8 weights, how do we generate 8 different 1024 bit kernels. It might have something to do with the wt_pre_sel signal and the sequence of atom operations. I guess I'll look into this further but it would be wonderful if you can provide some high-level intuitive guidance. Thank you so much!

silvaurus, 2018-05-17:
My understanding on pre-selection is that, for atomic and stripe operations, either the input data or the input weight need not to be changed, pre-selection can keep those data registered and selected. But I still don't understand those 'need to be changed' part, how do we convert a 128*8 bit input to 8 different 1024 processed input (which corresponds to 8 kernels), then dispatch them to 8 MAC cells.

Thanks!

Hassan313, 2020-09-09:
@silvaurus Hi, Thank you for the explanations. Have you figured out the answer to your questions?

Thank you.