nvdla/hw #139: Where to find detailed description of programming model

teabun, 2018-05-22:
I found on http://nvdla.org/hw/v1/hwarch.html#register-interface
that the register description is the brief version.

Where can I find the detailed description of these registers, and the fields within.

SCLUO, 2018-05-22:
You can find them in the RTL codes, in the single and double registers of each  main functional blocks.

teabun, 2018-05-22:
I am looking for a more detailed description of the programming model for each module from firmware perspective, to better understand the control of each module. So looking at the RTL isn't what I am looking for.

Thanks!

mmaciag, 2018-05-22:
Thumbs up to the questioner. Actual documentation of register interface is very concise. Some registers have bit fields which are not documented in architecture manual.

Of course, finding it out from RTL and figuring out the behavior or programming sequence is feasible, but it looks like unnecessary reverse engineering work, especially if there could be more detailed documentation. So this is actually a question about detailed documentation similar to _any decent microcontroller reference manual_.

@teabun If you look into `outdir/<model>/spec/manual` there are register definition files which include bit fields as well (opendla.xyz files). Some of them have brief descriptions.

SCLUO, 2018-05-22:
Agree. Let me revise my answer.  Definitions can be find in  [https://github.com/nvdla/hw/tree/master/spec/manual](url)
Although those are not yet as friendly as a manual.

nvhook, 2018-07-03:
The "Programming Guide" (http://nvdla.org/hw/v1/ias/programming_guide.html) includes additional details about registers and programming. Does it include the information you seek?