nvdla/hw #143: Using Questasim instead of VCS

Y159753, 2018-06-04:
I have the questasim license, not the vcs one. First, are the test benches absolutely necessary to test is the outdir was correctly built ?  
I see that the work to adapt questasim instead of vcs is consequent, VCS_HOME from tree.make is a variable spread widely into your code. I have very little experience in the field, do you think that the exchange of software might work ?

mlhwvip, 2018-07-18:
I also want to use questasim. Were you able to solve this issue?

Y159753, 2018-07-20:
I worked on other projects until now. But this is my task for the weeks to come. I will give updates soon. And I would be glad to collaborate with you.

nookfoo, 2018-08-23:
@Y159753 Do you have any update regarding alternative for VCS?

fnormandin-xeda, 2018-10-03:
Hi @Y159753, @nookfoo, @mlhwvip, 

Please come at @xtreme-eda and take a look!

We currently have a version of the **nv_small** branch working with **Questa 10.4b**.

Please be aware that this project seems very sensitive to tool versions. We tested our drop using Questa 10.4b, SystemC2.3.0 (from Questa installation), gcc4.9.2, python3.6, no DesignWare.

This is still work in progress, but as of now we are able to:
- Execute "./tools/bin/tmake -build ready_for_test" without error, which includes:
  - build the RTL model and all VIPs
  - build the SystemC reference model
  - build the trace generator testbench
  - build the trace player testbench (including linking the SystemC model and DPI)
- Run a trace generator base test (cd verif/testbench/trace_generator; make run)
- Run a trace player base test (cd verif/testbench/trace_player; make run)

We are currently working on the **verif_protection** target.

We are also working on supporting **Cadence IUS**.

Cheers!

smartchen2k, 2018-10-05:
what's the schedule time for Cadence IUS? @fnormandin-xeda

ycrad-xeda, 2018-10-05:
Hi @Y159753, @nookfoo, @mlhwvip, @smartchen2k,

The planned drop for Cadence should be pretty soon.
* Sometime next week, based on the fact that it took around 2 weeks to do perform the modifications and tests to get MTI/Questa working.

spatenaude-xeda, 2018-10-05:
In parallel we are now focusing on the coverage part in MTI, and of course similar later on for IUS. Tell us if you need something special.

fnormandin-xeda, 2018-10-05:
Hi @Y159753, @nookfoo, @mlhwvip,

Update:
We now have **verif_protection** target supported for Questa at @xtreme-eda.

[https://github.com/xtreme-eda/hw/tree/nv_small](url)

Working on **coverage** now.

thefpgaguy, 2018-10-18:
@fnormandin-xeda First of all thank you xtreame-eda for the great work. 

My company is getting a copy of QuestaSim. I understand that there are various options when purchasing QuestaSim. Are you able to share what would be the bare minimum requirements we need to get this simulation going?

Thanks in advance.

spatenaude-xeda, 2018-10-18:
Hello @thefpgaguy,
We appreciate that someone consider to use our contribution, thanks for that !
Once you populate our branch, you will see an extra file called README.XEDA
This one explains all the diffs we made to have Questa to run. Questa seems to accept sligthly different LRM coding style than VCS. In the process we discovered also that not all mix of SystemC/Questa/gcc(of questa)/gcc(of unix)/python were happy together. And to make it working we had to do some gymnastic (using gcc verX and pointing to a verY and so and so) this is not ideal, but this is reality check of this code. The version set we used is:
questa: 10.4b
python: 2.7 and 3 (per file...)
systemc: 2.3.0
gcc: gcc-4.7.4 (from Questa intall) + gcc-4.9.2 (from system)

In the case you need more help we, you can find us here: https://www.xtreme-eda.com/

Enjoy !

fnormandin-xeda, 2018-10-18:
Hello @thefpgaguy,

Thanks for your interest in our work.

Here is a list of required Questa licenses according to my observations:

msimcoverage
msimhdlsim
msimsystemc
msimviewer
mtiverification
svverification

That said, I don't know which minimum licensing package contains this subset. You may need to ask your Mentor representative...

In addition to @spatenaude-xeda answer, I wish to emphasize that our file set only supports Questa 10.4b. Especially, it will **not** run with version 10.6c.

thefpgaguy, 2018-10-19:
@spatenaude-xeda  @fnormandin-xeda Thanks so much for sharing the info. I'll check with our local Mentor rep.