nvdla/hw #144: Fail to generate vmod

Shayneyan, 2018-06-05:
I follow the instructions on website "http://nvdla.org/hw/v1/integration_guide.html#tree-build"
However, the outdir doesn't contain the vmod directory, while cmod and dmod exists.
The building log is shown in the below:

make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/spec/manual'
/home/ptzheng/jdk-10.0.1/bin/java -jar Ordt.jar -parms test.parms -systemverilog /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/sv/ -verilog /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_v.v -uvmregs /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_ral.sv -cppmod /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod -cppdrvmod /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod test.rdl
Open Register Design Tool, version=170915.01, input=test.rdl
Ordt: reading parameters from test.parms...
Ordt: building verilog...
Ordt: writing verilog file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_v.v...
Ordt: building systemverilog...
Ordt: writing systemverilog file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/sv/simple1_pio.sv...
Ordt: writing systemverilog file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/sv/simple1_jrdl_logic.sv...
Ordt: writing systemverilog file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/sv/simple1_jrdl_decode.sv...
Ordt: building UVM regs...
Ordt: writing UVM regs file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_ral.sv...
Ordt: building C++ model...
Ordt: writing C++ model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod/ordt_pio_common.hpp...
Ordt: writing C++ model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod/ordt_pio_common.cpp...
Ordt: writing C++ model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod/ordt_pio.hpp...
Ordt: writing C++ model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod/ordt_pio.cpp...
Ordt: building C++ driver model...
*** INFO ***: Overlay 0 total processed instances=16, unique instances=16, duplicate instances=0
Ordt: writing C++ driver model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod/ordt_pio_common.hpp...
Ordt: writing C++ driver model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod/ordt_pio_common.cpp...
Ordt: writing C++ driver model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod/ordt_pio_drv.hpp...
Ordt: writing C++ driver model file /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod/ordt_pio_drv.cpp...
Ordt complete Tue Jun 05 17:43:00 CST 2018
#@cp -f /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_v.v /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/; rm /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_v.v -rf
#@cp -f /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_ral.sv /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/; rm /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/regs_ral.sv -rf
#@cp -rf /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/sv/ /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/; rm /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/sv/ -rf
#@cp -rf /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/; rm /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/cmod -rf
#@cp -rf /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/; rm /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual/dmod -rf
==============================================
files are generated under /home/ptzheng/Downloads/hw-nvdlav1/outdir/test01/spec/manual
==============================================
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/spec/manual'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/spec/defs'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/spec/defs'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/vlibs'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/vlibs'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/include'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/include'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/rams/model'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/rams/model'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/rams/synth'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/rams/synth'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/apb2csb'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/apb2csb'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cdma'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cdma'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cbuf'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cbuf'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/csc'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/csc'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cmac'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cmac'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cacc'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cacc'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/sdp'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/sdp'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/pdp'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/pdp'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cdp'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/cdp'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/bdma'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/bdma'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/rubik'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/rubik'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/car'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/car'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/glb'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/glb'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/csb_master'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/csb_master'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/nocif'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/nocif'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/retiming'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/retiming'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/top'
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/vmod/nvdla/top'
make: Entering directory '/home/ptzheng/Downloads/hw-nvdlav1/verif/sim'
DUTDIR = dut
VIPDIR = vip
Using cleartext directories
/home/tools/vcs/mx-2015.09-SP2-9-T0426/bin/vcs -f /home/ptzheng/Downloads/hw-nvdlav1/verif/dut/dut.f  -Xddg=0x1 -o ./simv -cpp /home/utils/gcc-4.7.2/bin/g++  +nospecify +notimingchecks +define+VLIB_NO_UDP +warn=noTFIPC +warn=noTMR -full64 -y /home/ptzheng/Downloads/hw-nvdlav1/verif/synth_tb -y /home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/vlibs +incdir+/home/ptzheng/Downloads/hw-nvdlav1/verif/synth_tb +incdir+/home/ptzheng/Downloads/hw-nvdlav1/verif/dut +incdir+/home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/vlibs +incdir+/home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/include +incdir+/home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/vlibs +incdir+..  +define+DESIGNWARE_NOEXIST +vcs+lic+wait -sverilog +libext+.v +libext+.sv -timescale=1ns/1ns -debug_all +define+NVTOOLS_SYNC2D_GENERIC_CELL +define+NO_PERFMON_HISTOGRAM +define+PRAND_OFF  +define+NO_DUMPS /home/ptzheng/Downloads/hw-nvdlav1/verif/synth_tb [... cut ...] /hw-nvdlav1/verif/synth_tb/slave2mem_rd.v /home/ptzheng/Downloads/hw-nvdlav1/verif/synth_tb/slave2mem_wr.v /home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/vlibs/NV_DW02_tree.v /home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/vlibs/NV_DW_lsd.v /home/ptzheng/Downloads/hw-nvdlav1/verif/../outdir/nv_full/vmod/vlibs/NV_DW_minmax.v -l ./simv.compile.log   
Makefile:581: recipe for target 'simv' failed
make: Leaving directory '/home/ptzheng/Downloads/hw-nvdlav1/verif/sim'

nvhook, 2018-07-03:
@Shayneyan has this build failure persisted?

Shayneyan, 2018-07-05:
@nvhook No, this issue has been fixed by revising the default library path

shamalwinchurkar, 2018-12-30:
what is the default library path we need to set and where to fix?

killerzula, 2019-08-29:
@yankanghong what do you mean by revising the default library path? Kindly elaborate.