nvdla/hw #146: Verilator cannot be used for master branch

liuhengibm, 2018-06-06:
I can use the verilator testbench in v1 branch. But when I turned to the master branch and execute the ./tools/bin/tmake -build verilator command trying to build simulator, it generates the following log information: 

[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in vmod/vlibs 
[TMAKE]: building nv_small in vmod/include 
[TMAKE]: building nv_small in vmod/rams/model 
[TMAKE]: building nv_small in vmod/rams/synth 
[TMAKE]: building nv_small in vmod/nvdla/apb2csb 
[TMAKE]: building nv_small in vmod/nvdla/cdma 
[TMAKE]: building nv_small in vmod/nvdla/cbuf 
[TMAKE]: building nv_small in vmod/nvdla/csc 
[TMAKE]: building nv_small in vmod/nvdla/cmac 
[TMAKE]: building nv_small in vmod/nvdla/cacc 
[TMAKE]: building nv_small in vmod/nvdla/sdp 
[TMAKE]: building nv_small in vmod/nvdla/pdp 
[TMAKE]: building nv_small in vmod/nvdla/cfgrom 
[TMAKE]: building nv_small in vmod/nvdla/cdp 
[TMAKE]: building nv_small in vmod/nvdla/bdma 
[TMAKE]: building nv_small in vmod/nvdla/rubik 
[TMAKE]: building nv_small in vmod/nvdla/car 
[TMAKE]: building nv_small in vmod/nvdla/glb 
[TMAKE]: building nv_small in vmod/nvdla/csb_master 
[TMAKE]: building nv_small in vmod/nvdla/nocif 
[TMAKE]: building nv_small in vmod/nvdla/retiming 
[TMAKE]: building nv_small in vmod/nvdla/top 
[TMAKE]: building nv_small in verif/verilator 
[TMAKE]: DONE

This is the same log information when I executes ./tools/bin/tmake -build vmod. Seems it builds RTL, but does not build simulator. And when I continue to execute the "make run TEST=sanity0" command, the following error is reported:
make: *** No rule to make target `../../outdir/nv_small/verilator/test/sanity0/trace.bin', needed by `run'.  Stop.

So does verilator testbench work for master branch? Or we can only use vcs in master branch? @jwise @zdraw

mmaciag, 2018-06-06:
In master branch all Verilator tests for nv_small are in `<root>/verif/tests/trace_tests/nv_small`. So that would be something like: `make run TEST=dc_24x33x55_5x5x55x25_int8_0`. I tested them all and only few did not pass CRC check.

mmaciag, 2018-06-06:
Sorry, I am not maintainer/contributor so my knowledge is limited to what I already learned on this project. I work with Xilinx tools to run nv_small on FPGA and never touched vcs. AFAIK verilator tests are very limited when compared to vcs tests.

As for your second question, see: #145 - it seems that configurations other than nv_small are not yet ready to use.

terencebetter, 2018-07-13:
Hi, so you finished verilator test for nv_small model right? I'm building the verilator test now, but got some error like following: do you have any idea with that?

%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/nocif/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/nocif/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/nocif/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/pdp/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/pdp/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/pdp/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/retiming/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/retiming/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/retiming/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/rubik/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/rubik/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/rubik/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/sdp/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/sdp/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/sdp/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/top/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/top/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/top/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csb_master/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csb_master/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csb_master/RAMDP_8X66_GL_M1_E2.sv
%Error: Exiting due to too many errors encountered; --error-limit=50
%Error: Command Failed /usr/local/bin/verilator_bin --compiler clang --output-split 250000000 --cc --exe -f verilator_nv_small.f --Mdir ../../outdir/nv_small/verilator/ nvdla.cpp --compiler clang --output-split 250000000
make: *** [../../outdir/nv_small/verilator/VNV_nvdla.mk] Error 10
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

Thanks!

liuhengibm, 2018-07-13:
RAMDP_8X66_GL_M1_E2.v is not included in the verilator_nv_small.f filelist. You can add "-v ../../outdir/nv_small/vmod/rams/model/RAMDP_8X66_GL_M1_E2.v" to verif/verilator/verilator_nv_small.f and try again.

terencebetter, 2018-07-13:
Problem solved! Thanks!

zhanjf, 2019-03-14:
after using command ./tools/bin/tmake -build verilator
I tried to make run TEST=dc_24x33x55_5x5x55x25_int8_0 under hw director,

but I got the following log
make run TEST=dc_24x33x55_5x5x55x25_int8_0
make: *** No rule to make target 'run'.  Stop.

it seem no run target. So where can I run 
make run TEST=dc_24x33x55_5x5x55x25_int8_0?
Thanks!

Hong-333, 2019-04-17:
Hi I solved the same problem as above, but I have another problem

%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_rd_weight_bdma
%Warning-PINMISSING: Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_rd_weight_rbk
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_rd_weight_sdp_e
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_wr_weight_bdma
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_wr_weight_rbk
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_hybrid
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_int_data
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_le_hit
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_lo_hit
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_oflow
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_uflow
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_algo
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_offset
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_scale
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_truncate
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_operand
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_src
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_lut_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_offset
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_scale
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_truncate
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_operand
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_prelu
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_src
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_truncate
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_hybrid_priority
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_access_type
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_addr
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_data
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_data_wr
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_table_id
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_end
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_function
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_index_offset
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_index_select
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_slope_oflow_scale
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_slope_oflow_shift
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_slope_uflow_scale
[... cut ...]
%Warning-IMPLICIT: ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:338: Signal definition not found, creating implicitly: nvdla_core2cvsram_r_rready
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/local/bin/verilator_bin --compiler clang --output-split 250000000 --cc --exe -f verilator_nv_small.f --Mdir ../../outdir/nv_small/verilator/ nvdla.cpp --compiler clang --output-split 250000000
make: *** [../../outdir/nv_small/verilator/VNV_nvdla.mk] Error 10
[TMAKE]: DIE when building nv_small verif/verilator at ./tools/bin/tmake line 278.
[TMAKE]: nv_small: FAIL

qaziullah, 2021-04-14:
Hi  @liuhengibm and @terencebetter 
[files.zip](https://github.com/nvdla/hw/files/6308574/files.zip)

 I am also getting error for running ./tools/bin/tmake -build verilator. I have added "-v ../../outdir/nv_small/vmod/rams/model/RAMDP_8X66_GL_M1_E2.v" to the verilator_nv_small.f file.  the file can be found [here](https://file.io/7Jg0svFxiu74).

Starting few lines of the nv_small_verilator.log file are shared below. Full file can be found [here](https://file.io/BdvHLq0G5Hp0).

command: cd verif/verilator;make PROJECT=nv_small 
/usr/bin/verilator --compiler clang --output-split 250000000 --cc --exe -f verilator_nv_small.f --Mdir ../../outdir/nv_small/verilator/ nvdla.cpp --compiler clang --output-split 250000000
%Warning-REDEFMACRO: ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:35: Redefining existing define: 'DESIGNWARE_NOEXIST', with different value: 1
                     ... Use "/* verilator lint_off REDEFMACRO */" and lint_on around source to disable this message.
%Warning-REDEFMACRO: Previous definition is here, with value: 
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: 'reg2dp_rd_weight_bdma'
NV_NVDLA_MCIF_csb u_csb (
                  ^~~~~
                     ../../outdir/nv_small/vmod/nvdla/top/NV_NVDLA_partition_o.v:557: ... note: In file included from NV_NVDLA_partition_o.v
                     ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:345: ... note: In file included from NV_nvdla.v
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: 'reg2dp_rd_weight_rbk'
NV_NVDLA_MCIF_csb u_csb (
                  ^~~~~
                     ../../outdir/nv_small/vmod/nvdla/top/NV_NVDLA_partition_o.v:557: ... note: In file included from NV_NVDLA_partition_o.v
                     ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:345: ... note: In file included from NV_nvdla.v
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: 'reg2dp_rd_weight_sdp_e'
NV_NVDLA_MCIF_csb u_csb (
                  ^~~~~
                     ../../outdir/nv_small/vmod/nvdla/top/NV_NVDLA_partition_o.v:557: ... note: In file included from NV_NVDLA_partition_o.v
                     ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:345: ... note: In file included from NV_nvdla.v
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: 'reg2dp_wr_weight_bdma'
NV_NVDLA_MCIF_csb u_csb (
                  ^~~~~
                     ../../outdir/nv_small/vmod/nvdla/top/NV_NVDLA_partition_o.v:557: ... note: In file included from NV_NVDLA_partition_o.v
                     ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:345: ... note: In file included from NV_nvdla.v
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: 'reg2dp_wr_weight_rbk'