nvdla/hw #148: Sim Makefile Path Not Found

silvaurus, 2018-06-07:
Hi! I'm running the makefile under `hw/verif/sim`, by just doing `make build`, `make run TESTDIR=../traces/traceplayer/conv_8x8_fc_int16/`, and it gives me a couple of warnings relating to path not found. I run make file under `hw/verif/sim` directory.

Cannot open ../../synth_tb/syn_tb_defines.vh at /mnt/y-lab/henry/hw/verif/synth_tb/sim_scripts/inp_txn_to_hexdump.pl line 304.
Warning-[STASKW_CO1] Cannot open file
  The file 'input.txn.raw' could not be opened. No such file or directory.
  Please ensure that the file exists with proper permissions.

Warning-[STASKW_RMCOF] Cannot open file
/mnt/y-lab/henry/hw/verif/synth_tb/csb_master_seq.v, 110
  Cannot open file 'input.txn.raw' passed as argument to $readmem.
  Please verify that the first argument to $readmem is a file that exists with
  proper permissions.

silvaurus, 2018-06-07:
I think this issue only occurs on `conv_8x8_fc_int16`

jmeng1988, 2018-06-25:
Hi, 
the path for "syn_tb_defines.vh" is wrong. you can run just modify line 56 in synth_tb/sim_scripts/inp_txn_to_hexdump.pl as the following.

populate_mseq_defines("../synth_tb/syn_tb_defines.vh", \%MSEQ_defines);

after that, you should can run the sims.