nvdla/hw #156: error while executing the command: tools/bin/tmake -build cmod_top

Adithyak1998, 2018-06-16:
I got this output when i executed the command: tools/bin/tmake -build cmod_top

make: Entering directory '/root/vp/hw/spec/defs'
/home/utils/gcc-4.9.3/bin/cpp -undef -nostdinc -P -C nv_full.spec -o /root/vp/hw/outdir/nv_full/spec/defs/project.def
make: /home/utils/gcc-4.9.3/bin/cpp: Command not found
make: *** [/root/vp/hw/outdir/nv_full/spec/defs/project.def] Error 127
Makefile:22: recipe for target '/root/vp/hw/outdir/nv_full/spec/defs/project.def' failed
make: Leaving directory '/root/vp/hw/spec/defs'
make: Entering directory '/root/vp/hw/spec/manual'
/home/utils/java/jdk1.8.0_131/bin/java -jar Ordt.jar -parms test.parms -systemverilog /root/vp/hw/outdir/nv_full/spec/manual/sv/ -verilog /root/vp/hw/outdir/nv_full/spec/manual/regs_v.v -uvmregs /root/vp/hw/outdir/nv_full/spec/manual/regs_ral.sv -cppmod /root/vp/hw/outdir/nv_full/spec/manual/cmod -cppdrvmod /root/vp/hw/outdir/nv_full/spec/manual/dmod test.rdl
make: /home/utils/java/jdk1.8.0_131/bin/java: Command not found
make: *** [default] Error 127
Makefile:20: recipe for target 'default' failed
make: Leaving directory '/root/vp/hw/spec/manual'
make: Entering directory '/root/vp/hw/cmod'
mkdir -p ../outdir/nv_full/cmod/../cmod/nvdla_payload/
/home/utils/gcc-4.9.3/bin/g++ -I./include -I../cmod/include/nvdla_ness_header -I../cmod/include -I../cmod/nvdla_payload -I../cmod/bdma -I../cmod/cacc -I../cmod/cbuf -I../cmod/cdma -I../cmod/cdp -I../cmod/cmac -I../cmod/csb_master -I../cmod/csc -I../cmod/cvif -I../cmod/glb -I../cmod/hls_wrapper -I../cmod/mcif -I../cmod/nvdla_core -I../cmod/pdp -I../cmod/rubik -I../cmod/sdp -I../cmod/bdma/gen -I../cmod/cacc/gen -I../cmod/cbuf/gen -I../cmod/cdma/gen -I../cmod/cdp/gen -I../cmod/cmac/gen -I../cmod/csb_master/gen -I../cmod/csc/gen -I../cmod/cvif/gen -I../cmod/glb/gen -I../cmod/hls_wrapper/gen -I../cmod/mcif/gen -I../cmod/nvdla_core/gen -I../cmod/pdp/gen -I../cmod/rubik/gen -I../cmod/sdp/gen -I../cmod/hls/include -I../cmod/hls/cdma_libs -I../cmod/hls/cdp_libs -I../cmod/hls/csc_libs -I../cmod/hls/sdp -I../cmod/hls/vlibs -I/usr/local/systemc-2.3.0//include -MMD -MP -fPIC -Wall -Werror -DSC_INCLUDE_DYNAMIC_PROCESSES -Wp,-w -std=c++11 -c ../cmod/nvdla_payload/dla_b_transport_payload.cpp -o ../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o
make: /home/utils/gcc-4.9.3/bin/g++: Command not found
make: *** [../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o] Error 127
Makefile:165: recipe for target '../outdir/nv_full/cmod/../cmod/nvdla_payload/dla_b_transport_payload.cpp.o' failed
make: Leaving directory '/root/vp/hw/cmod'
logfile: outdir/build.log
Filehandle GEN1 opened only for input at /usr/share/perl5/IO/Tee.pm line 132.
==================BUILD PASS==================
Filehandle GEN1 opened only for input at /usr/share/perl5/IO/Tee.pm line 132.

Filehandle GEN1 opened only for input at /usr/share/perl5/IO/Tee.pm line 132.

These are the addresses included in my PATH variable:
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/utils/java/jdk1.8.0_131/bin:/home/utils/gcc-4.9.3/bin

Please help me in identifying the problem.
Thanks

sun3388, 2018-06-18:
Hi 
Could you help try remove the default path defined in https://github.com/nvdla/hw/blob/nvdlav1/Makefile

DEFAULT_CPP  := /home/utils/gcc-4.9.3/bin/cpp
DEFAULT_GCC  := /home/utils/gcc-4.9.3/bin/g++
DEFAULT_PERL := /home/utils/perl-5.8.8/bin/perl
DEFAULT_JAVA := /home/utils/java/jdk1.8.0_131/bin/java
DEFAULT_SYSTEMC := /usr/local/systemc-2.3.0/
DEFAULT_VERILATOR := verilator
DEFAULT_CLANG     := clang
DEFAULT_PROJ := nv_full

it is assuming your tool is defined in /home/utils/... 's path

nvhook, 2018-07-03:
Duplicate of nvdla/vp issue #13, closing.

closed 2018-07-03 by nvhook