nvdla/hw #157: How to use cmod code for simulation

wxbbuaa2011, 2018-06-21:
we can't find C simulation related documents, so how do we use C code for simulation verification? Thank you!

nvdsmith, 2018-06-23:
The testbench located in the master branch also uses the C model for verification.  You can take a look at this testbench to see how it's called.  uvmconnect is used to link in the c model to the verilog simulation.

ferin08, 2018-07-02:
@JunningWu can we use the VP complied outputs for the NVDLA hardware implementation ?

nvdsmith, 2018-07-02:
The virtual platform utilizes the same C model.  When you build the virtual platform, one of the steps is:

3. Download and build NVDLA CMOD

This is the same code and build step as used for the RTL verification environment. So as long as you're using the same local branch, you can share the compiled results between the two.