nvdla/hw #158: Error when build verif_trace_pleyer

dai-pch, 2018-06-21:
I got an error when build verif_trace_player using commands:
./tools/bin/tmake -only verif_trace_player 

The output is as follows:
[TMAKE]: building nv_small in verif/testbench/trace_player 
Died at ./tools/bin/tmake line 254.
make: *** [sysc_build] Error 1
[TMAKE]: DONE

Error log in outdir/nv_small.verif_trace_player.log is:

command: cd verif/testbench/trace_player;make PROJECT=nv_small 
/bin/mkdir -p ../../../outdir/nv_small/verif/testbench/trace_player/ && cd ../../../outdir/nv_small/verif/testbench/trace_player/ && \
	syscan -sysc=230 -V -work DEFAULT -l sysc.log \
	 -full64 -cc gcc-4.7.2 -cpp g++-4.7.2 -tlm2 -sysc=opt_if  -cflags " -D__STDC_FORMAT_MACROS -DNVDLA_REFERENCE_MODEL_ENABLE  -DUVMC_MAX_WORDS=2048 -DVCS -I/home/pengcheng/projects/nvdla/hw/third_party_tools/uvmc-2.3.0/src/connect/sc -I/opt/EDA/synopsys/vcs/J-2014.12-SP3/include/systemc23 -I/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/release/include -I/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_scsv/sc -I/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc" /home/pengcheng/projects/nvdla/hw/third_party_tools/uvmc-2.3.0/src/connect/sc/uvmc.cpp /home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_scsv/sc/nvdla_dbb_scsv_extension_packer.cpp /home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_scsv/sc/nvdla_scsv_converter.cpp /home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_layer.cpp /home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/log.cpp /home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp:nvdla_top_sc_adapter -Mdir=csrc

Note-[SC-SYSCAN-USE-CC] Using C-compiler
  Using 'gcc-4.7.2' C-compiler for C compilation.

Note-[SC-SYSCAN-USE-CPP] Using C++-compiler
  Using 'g++-4.7.2' C++-compiler for C++ compilation.

gmake[1]: Nothing to be done for `module-uvmc'.
gmake[1]: Nothing to be done for `module-nvdla_dbb_scsv_extension_packer'.
gmake[1]: Nothing to be done for `module-nvdla_scsv_converter'.
gmake[1]: Nothing to be done for `module-nvdla_top_sc_layer'.
gmake[1]: Nothing to be done for `module-log'.
Analyzing the interface of /home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp
"/home/pengcheng/projects/nvdla/hw/third_party_tools/uvmc-2.3.0/src/connect/sc/uvmc_convert.h", line 288: error: 
          class "uvmc_converter<tlm::tlm_generic_payload>" is not an entity
          that can be instantiated
  template class uvmc_converter<tlm_generic_payload>;
           ^

"/home/pengcheng/projects/nvdla/hw/third_party_tools/uvmc-2.3.0/src/connect/sc/uvmc_convert.h", line 366: error: 
          class "uvmc_print<tlm::tlm_generic_payload>" is not an entity that
          can be instantiated
  template class uvmc_print<tlm_generic_payload>;
           ^

"/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp", line 39: error: 
          identifier "nvdla_top_sc_inst" is undefined
    nvdla_top_sc_inst = new scsim::cmod::NV_nvdla("nvdla_top_sc_inst");
    ^

"/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp", line 39: error: 
          name followed by "::" must be a class or namespace name
    nvdla_top_sc_inst = new scsim::cmod::NV_nvdla("nvdla_top_sc_inst");
                            ^

"/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp", line 39: error: 
          expected a type specifier
    nvdla_top_sc_inst = new scsim::cmod::NV_nvdla("nvdla_top_sc_inst");
                            ^

"/home/pengcheng/projects/nvdla/hw/outdir/nv_small/verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp", line 44: error: 
          identifier "nvdla_top_sc_layer_inst" is undefined
    nvdla_top_sc_layer_inst = new nvdla_top_sc_layer("nvdla_top_sc_layer_inst"); 
    ^

Error-[SC-SCAN-SCFE] Cannot read the SystemC database
  Cannot read the SystemC database 
  '/home/pengcheng/nvdla-hw/outdir/nv_small/verif/testbench/trace_player/csrc/sysc/nvdla_top_sc_adapter/nvdla_top_sc_adapter.scfe',
  encountering message 'Could not open file 
  /home/pengcheng/nvdla-hw/outdir/nv_small/verif/testbench/trace_player/csrc/sysc/nvdla_top_sc_adapter/nvdla_top_sc_adapter.scfe'.
  Please make sure that the file exists and you have read permission, then try
  again.

systemc2vdef failed
make: *** [sysc_build] Error 1

I'm using gcc 4.7.2, systemc-2.3.0. Can anyone help me?

dai-pch, 2018-06-25:
That means using a new version of vcs would be ok? I'll try it later.

dai-pch, 2018-06-25:
I've figure out that there are two problems.
First one is in file third_party_tools/uvmc-2.3.0/src/connect/sc/uvmc_convert.h.
There are two nonstandard syntax at line 288 and 366 respectively. Key word "template" in the begin of these lines should be removed. (Of course these are not problems of this project.)

Second one is in file /verif/vip/reference_model/nvdla_top_scsv_adapter/sc/nvdla_top_sc_adapter.cpp, from line 40 to line 119.
The error said identifiers nvdla_top_sc_inst and nvdla_top_sc_layer_inst are not defined. 
I find they are disabled by macor "#ifndef __EDG__" in nvdla_top_sc_adapter.h and nvdla_top_sc_layer.h. So I disabled these lines in the same way.

When I do these changes, it can be built without error. I don't know whether it's right or not to do this. Or will this changes impact other parts. I think the second problem may be a bug. Please check that.

smartchen2k, 2018-10-09:
@dai-pch 
I'm using vcs 2016, no error occurs when runing ./tools/bin/tmake -only verif_trace_player .
So, maybe the modification is  just fixed your problem by coincidence.
For reference only.