nvdla/hw #160: Hardware Architectural Specification about registers？

wxbbuaa2011, 2018-06-25:
I didn't find many specific meanings of registers.such as D_MISC_CFG ，0x5014 ， Configuration of operation: convolution mode, precision, weight reuse, data reuse. The register is set by CDMA module. And I see write_reg 0xffff1405 0x1001100 #NVDLA_CDMA.D_MISC_CFG_0 in .\hw-nvdlav1\verif\traces\traceplayer\googlenet_conv2_3x3_int16\input.txn. But I don't know what is specifically configured. Is there related documentation in this field? Thank you very much.

mmaciag, 2018-06-25:
I'm afraid things did not change much since issue #139. You will have to look inside `spec/manual`, directly inside HDL code or wait until documentation update.

wxbbuaa2011, 2018-06-25:
Thank you very much. I have an other question. I can't find a CBUF module example such as input.txn file. Can you help me? Thanks.