nvdla/hw #168: Question: Why implementation scheme of NV_NVDLA_CDMA_WG_fifo differs from others?

mmaciag, 2018-06-29:
The NV_NVDLA_CDMA_WG_fifo in contrast to other FIFOs in the design uses 'folded RAM'. It looks like it is wrapping ordinary RAM with different depth and width than upper module with some glue logic. Could anyone help me to understand why it is so? What NV_NVDLA_CDMA_WG_fifo_folded_ram_rws_128x5 tries to achieve?

nvhook, 2018-06-29:
Tall skinny FIFOs make inefficient use of power because they must charge/access long bit lines in RAMs. If the RAM is <= 5 bits wide, it is also inefficient in area. Folding decreases the underlying RAM depth by a power-of-2 and increases the underlying RAM width by the same power-of-2.

closed 2018-07-02 by nvhook

mmaciag, 2018-07-03:
Sorry for delayed answer and not closing issue. Thank you, now it is clear.