nvdla/hw #172: change 'spec' file doesn't change RTL code

terencebetter, 2018-07-06:
Hi,
I know that if I change the spec file parameters I can get different NVDLA module. But, I found, it will not change the code in 'outdir' after I build vmod. Is there any way can modify the NVDLA modue?

Thanks!

mmaciag, 2018-07-06:
The `outdir` is a result of "code generation" system invented by NVDLA team (a mixture of Perl and C++ preprocessing magic). After changing spec file you have to run ./tools/bin/tmake -build again. This is because C++ preprocessor brutally removes parts of unused code, and Perl generates a lot of boilerplate code. If you applied some modifications to outdir, you would need to integrate them to `vmod` first.

terencebetter, 2018-07-07:
Hi, the 'outdir' comes up after I type "./tools/bin/tmake -build vmod" . So do you meaning I need to type it again?

And I just found, when I disable the BDMA parts, the system will not pass the first test "sanity0" does it mean, I had already turn off the BDMA part? 

Thanks for your help!

mmaciag, 2018-07-09:
I guess you are working on *nvdlav1* branch? I am not sure how much this branch is configurable. The sanity tests might not work with other specs than *nv_full* as it is. But in general after changing anything in .spec file, you have to run that command again.

terencebetter, 2018-07-09:
Yeah, thanks for your help!

terencebetter, 2018-07-10:
Hi, I'm just wondering why there is no nv_small model? Since, I want porting a small NVDLA model into my FPGA, and nv_full seems too large, I'm looking for the small model RTL code, but after I change the spec file and generate the RTL code, nothing seems changed, although 'define' file is changed as I did in spec file. 

Thanks!

sun3388, 2018-07-10:
Maybe you can refer to the AWS FPGA demo(http://nvdla.org/vp_fpga.html), which is using the nv_small

mmaciag, 2018-07-11:
```
git checkout master
```

terencebetter, 2018-07-11:
I got it! Thanks!

closed 2018-07-11 by terencebetter