nvdla/hw #174: Build NVDLA large for 1024MAC

m1kezh, 2018-07-11:
Hi,

I tried to configure and build a 1024 MAC NVDLA large RTL, as change the following parameters in nv_large.spec, and let the other parameters the same as default.
#define MAC_ATOMIC_C_SIZE_32
#define MAC_ATOMIC_K_SIZE_32

While running ./tools/bin/tmake -build vmod, I get a error and tmake building RTL failed, the log is as following, while changing MAC_ATOMIC_C_SIZE to 64, it can build successfully. Does NVDLA large mode not support 1024 MAC? or are there some other parameters in spec should also be set?

Thanks!

[zhouhengzhao@bj-idc1-10-10-36-29 hw-master]$ ./tools/bin/tmake -build vmod
[TMAKE]: building nv_large_zhouhzh in spec/defs 
[TMAKE]: building nv_large_zhouhzh in spec/manual 
[TMAKE]: building nv_large_zhouhzh in spec/odif 
[TMAKE]: building nv_large_zhouhzh in vmod/vlibs 
[TMAKE]: building nv_large_zhouhzh in vmod/include 
[TMAKE]: building nv_large_zhouhzh in vmod/rams/model 
[TMAKE]: building nv_large_zhouhzh in vmod/rams/synth 
[TMAKE]: building nv_large_zhouhzh in vmod/rams/fpga/small_rams 
[TMAKE]: building nv_large_zhouhzh in vmod/nvdla/apb2csb 
[TMAKE]: building nv_large_zhouhzh in vmod/nvdla/cdma 
Processing NV_NVDLA_CDMA_CVT_cell.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_CVT_cell.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_CVT_cell.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_CVT_cell.v unchanged.
Processing NV_NVDLA_CDMA_DC_fifo.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_DC_fifo.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_DC_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_DC_fifo.v unchanged.
Processing NV_NVDLA_CDMA_IMG_ctrl.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_ctrl.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_ctrl.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_ctrl.v unchanged.
Processing NV_NVDLA_CDMA_IMG_fifo.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_fifo.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_fifo.v unchanged.
Processing NV_NVDLA_CDMA_IMG_pack.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_pack.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_pack.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_pack.v unchanged.
Processing NV_NVDLA_CDMA_IMG_sg.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_sg.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_sg.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_sg.v unchanged.
Processing NV_NVDLA_CDMA_IMG_sg2pack_fifo.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_sg2pack_fifo.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_sg2pack_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_IMG_sg2pack_fifo.v unchanged.
Processing NV_NVDLA_CDMA_WG_fifo.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WG_fifo.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WG_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WG_fifo.v unchanged.
Processing NV_NVDLA_CDMA_WT_fifo.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_fifo.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_fifo.v unchanged.
Processing NV_NVDLA_CDMA_WT_sp_arb.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_sp_arb.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_sp_arb.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_sp_arb.v unchanged.
Processing NV_NVDLA_CDMA_WT_wgs_fifo.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_wgs_fifo.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_wgs_fifo.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_wgs_fifo.v unchanged.
Processing NV_NVDLA_CDMA_WT_wrr_arb.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_wrr_arb.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_wrr_arb.v.vcp successfully written
eperl: ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_WT_wrr_arb.v unchanged.
Processing NV_NVDLA_CDMA_cvt.v into ../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_cvt.v.vcp
../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_cvt.v.vcp successfully written
ERROR: eperl: eval:
&eperl::flop("-nodeclare   -rval \"1'b0\"                                   -d \"cvt_wr_en\"             -q cvt_wr_en_d1");
&eperl::flop("-nodeclare   -rval \"1'b0\"        -en \"cvt_wr_en\"          -d \"cvt_wr_mean\"           -q cvt_wr_mean_d1");
&eperl::flop("-nodeclare   -rval \"1'b0\"        -en \"cvt_wr_en\"          -d \"cvt_wr_uint\"           -q cvt_wr_uint_d1");
&eperl::flop("-nodeclare  -norst                 -en \"cvt_wr_en & cvt_wr_mean\" -d \"cvt_wr_mean_data\" -q cvt_wr_mean_data_d1");
&eperl::flop("-nodeclare  -norst                 -en \"cvt_wr_en\"          -d \"cvt_wr_data\"           -q cvt_wr_data_d1");
my $dmaif=512;
my $bpe = 8;
my $Bnum = $dmaif / $bpe;
&eperl::flop("-nodeclare   -rval \"${Bnum}'b0\"           -en \"cvt_wr_en | cvt_wr_en_d1\" -d \"cvt_cell_en\"     -q cvt_cell_en_d1");
&eperl::flop("-nodeclare   -rval \"1'b0\"                                   -d \"cvt_out_vld\"           -q cvt_out_vld_d1");
&eperl::flop("-nodeclare   -rval \"1'b0\"                                   -d \"img2cvt_dat_wr_en\"     -q cvt_out_pad_vld_d1");
&eperl::flop("-nodeclare   -rval \"{17{1'b0}}\"  -en \"cvt_wr_en\"          -d \"cvt_out_addr\"          -q cvt_out_addr_d1");
&eperl::flop("-nodeclare   -rval \"{4{1'b0}}\"   -en \"cvt_wr_en\"          -d \"cvt_wr_mask\"           -q cvt_out_nz_mask_d1");
&eperl::flop("-nodeclare   -rval \"${Bnum}'b0\"           -en \"img2cvt_dat_wr_en\"  -d \"cvt_wr_pad_mask\"       -q cvt_out_pad_mask_d1");
my $atmc=32*8;
my $k = int(log(int($atmc/$dmaif))/log(2));
if($dmaif < $atmc) {
&eperl::flop("-nodeclare   -rval \"${k}'b0\"                                      -d \"cvt_out_reg_en\"        -q cvt_out_reg_en_d1");
&eperl::flop("-nodeclare   -rval \"{2{1'b0}}\"   -en \"cvt_wr_en\"          -d \"cvt_out_sel\"           -q cvt_out_sel_d1");
} else {
&eperl::flop("-nodeclare   -rval \"1'b0\"                                      -d \"cvt_out_reg_en\"        -q cvt_out_reg_en_d1");
}

error:
Can't take log of 0 at (eval 22) line 16, <IN1> line 617.

 at ../../../tools/bin/eperl line 265
	eperl_main::Error('eval:\x{a}&eperl::flop("-nodeclare   -rval \"1\'b0\"             ...') called at ../../../tools/bin/eperl line 210
	eperl_main::EvalScript('&eperl::flop("-nodeclare   -rval \"1\'b0\"                   ...') called at ../../../tools/bin/eperl line 189
	eperl_main::ProcessInput('../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDM...', '../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDM...') called at ../../../tools/bin/eperl line 158
make: *** [../../../outdir/nv_large_zhouhzh/vmod/nvdla/cdma/NV_NVDLA_CDMA_cvt.v] Error 25
Died at ./tools/bin/tmake line 255.
[TMAKE]: DONE

nvhook, 2018-07-24:
Maybe you've already figured this out, but it looks like you have a typo with the underscore between SIZE and 32: 
#define MAC_ATOMIC_C_SIZE_32
#define MAC_ATOMIC_K_SIZE_32

Then the error message "Can't take log of 0" would be explained by this: 
define NVDLA_MAC_ATOMIC_C_SIZE_LOG2    log2(NVDLA_MAC_ATOMIC_C_SIZE)

Problem resolved?

m1kezh, 2018-07-25:
MAC_ATOMIC_C_SIZE should be defined as MAC_ATOMIC_C_SIZE_8 /  MAC_ATOMIC_C_SIZE_32 / MAC_ATOMIC_C_SIZE_64, it is a requirement as NVDLA spec syntax, otherwise you will get a error like this: 
projects.spec:132:6: error: #error "one of NVDLA_MAC_ATOMIC_C_SIZE_{64,32,8} must be set"
     #error "one of NVDLA_MAC_ATOMIC_C_SIZE_{64,32,8} must be set"

I tried MAC_ATOMIC_C_SIZE_8, MAC_ATOMIC_C_SIZE_32, MAC_ATOMIC_C_SIZE_64, only MAC_ATOMIC_C_SIZE_32 can build vmod successful.  

You cannot remove the underscore between SIZE and 32, you will get the same error:
In file included from nv_large.spec:44:0:
projects.spec:132:6: error: #error "one of NVDLA_MAC_ATOMIC_C_SIZE_{64,32,8} must be set"
     #error "one of NVDLA_MAC_ATOMIC_C_SIZE_{64,32,8} must be set"
      ^
make: *** [/home/zhouhengzhao/work/NVDLA/v2_large/hw-master/outdir/nv_large/spec/defs/project.def] Error 1
Died at ./tools/bin/tmake line 255.