nvdla/hw #175: nv_small (master branch) verilator building error

terencebetter, 2018-07-13:
Hi, I'm trying to build the verilator test for nv_small model, I can successfully build RTL code by running 
./tools/bin/tmake -build vmod

But, when I trying to build verilator test by running
./tools/bin/tmake -build verilator
I got those error: 

%Warning-WIDTH: ../../outdir/nv_small/vmod/vlibs/NV_NVDLA_HLS_shiftrightsu.v:42: Operator ADD expects 10 bits on the RHS, but RHS's VARREF 'point5' generates 1 bits.
%Warning-WIDTH: ../../outdir/nv_small/vmod/vlibs/NV_NVDLA_HLS_shiftrightsu.v:47: Operator GTE expects 32 bits on the LHS, but LHS's VARREF 'shift_num' generates 5 bits.
%Warning-WIDTH: ../../outdir/nv_small/vmod/vlibs/NV_NVDLA_HLS_shiftrightsu.v:42: Operator ADD expects 9 bits on the RHS, but RHS's VARREF 'point5' generates 1 bits.
%Warning-WIDTH: ../../outdir/nv_small/vmod/vlibs/NV_NVDLA_HLS_shiftrightsu.v:47: Operator GTE expects 32 bits on the LHS, but LHS's VARREF 'shift_num' generates 6 bits.
%Warning-WIDTH: ../../outdir/nv_small/vmod/vlibs/NV_NVDLA_HLS_shiftrightsu.v:42: Operator ADD expects 33 bits on the RHS, but RHS's VARREF 'point5' generates 1 bits.
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348: Cannot find file containing module: RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348: Looked in:
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/bdma/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/bdma/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/bdma/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cacc/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cacc/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cacc/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/car/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/car/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/car/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cbuf/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cbuf/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cbuf/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cdma/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cdma/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cdma/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cdp/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cdp/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cdp/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cmac/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cmac/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/cmac/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csc/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csc/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csc/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/glb/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/glb/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/glb/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/nocif/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/nocif/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/nocif/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/pdp/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/pdp/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/pdp/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/retiming/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/retiming/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/retiming/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/rubik/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/rubik/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/rubik/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/sdp/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/sdp/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/sdp/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/top/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/top/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/top/RAMDP_8X66_GL_M1_E2.sv
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csb_master/RAMDP_8X66_GL_M1_E2
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csb_master/RAMDP_8X66_GL_M1_E2.v
%Error: ../../outdir/nv_small/vmod/rams/synth/nv_ram_rwsp_8x65_logic.v:348:       ../../outdir/nv_small/vmod/nvdla/csb_master/RAMDP_8X66_GL_M1_E2.sv
%Error: Exiting due to too many errors encountered; --error-limit=50
%Error: Command Failed /usr/local/bin/verilator_bin --compiler clang --output-split 250000000 --cc --exe -f verilator_nv_small.f --Mdir ../../outdir/nv_small/verilator/ nvdla.cpp --compiler clang --output-split 250000000
make: *** [../../outdir/nv_small/verilator/VNV_nvdla.mk] Error 10
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

I can successfully run verilator test in nv_full model in Nvdlav1. Please help me solve this.

Thanks!

terencebetter, 2018-07-13:
Problem solved! Thanks!

xivisi, 2018-07-13:
how to ?

mmaciag, 2018-07-13:
I guess @terencebetter used RAMs from _fpga_ folder not _synth_.

terencebetter, 2018-07-13:
refer to the answer in https://github.com/nvdla/hw/issues/146#issuecomment-404773479

mmaciag, 2018-07-13:
Right... didn't notice further comments there. By the way RAMs from _fpga_ can be used as well, if there is such need (it could maybe increase the simulation speed). Yet necessity of modifying _.f_ file is still valid.

closed 2018-07-23 by nvhook

Hong-333, 2019-04-17:
Hi I solved the same problem as above, but I have another problem

%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_rd_weight_bdma
%Warning-PINMISSING: Use "/* verilator lint_off PINMISSING */" and lint_on around source to disable this message.
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_rd_weight_rbk
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_rd_weight_sdp_e
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_wr_weight_bdma
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/nocif/NV_NVDLA_mcif.v:265: Cell has missing pin: reg2dp_wr_weight_rbk
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_hybrid
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_int_data
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_le_hit
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_lo_hit
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_oflow
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: dp2reg_lut_uflow
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_algo
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_offset
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_scale
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_cvt_truncate
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_operand
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_alu_src
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_lut_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_bypass
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_offset
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_scale
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_cvt_truncate
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_operand
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_prelu
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_mul_src
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_ew_truncate
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_hybrid_priority
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_access_type
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_addr
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_data
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_data_wr
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_int_table_id
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_end
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_function
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_index_offset
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_index_select
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_slope_oflow_scale
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_slope_oflow_shift
%Warning-PINMISSING: ../../outdir/nv_small/vmod/nvdla/sdp/NV_NVDLA_sdp.v:365: Cell has missing pin: reg2dp_lut_le_slope_uflow_scale
[... cut ...]
%Warning-IMPLICIT: ../../outdir/nv_small/vmod/nvdla/top/NV_nvdla.v:338: Signal definition not found, creating implicitly: nvdla_core2cvsram_r_rready
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/local/bin/verilator_bin --compiler clang --output-split 250000000 --cc --exe -f verilator_nv_small.f --Mdir ../../outdir/nv_small/verilator/ nvdla.cpp --compiler clang --output-split 250000000
make: *** [../../outdir/nv_small/verilator/VNV_nvdla.mk] Error 10
[TMAKE]: DIE when building nv_small verif/verilator at ./tools/bin/tmake line 278.
[TMAKE]: nv_small: FAIL