nvdla/hw #176: Depth-wise Convolution on NVDLA

kchhavi, 2018-07-16:
Hi,

Can NVDLA do depth-wise convolution (as used in Mobilenet) ? 
If yes, how can the RTL be configured for it?

Thanks.

kchhavi, 2018-07-24:
I could program NVDLA h/w for depth-wise convolution and I observe very low MAC utilization with it (using one multiplier only). I have read that using winograd with depth-wise convolution can improve the MAC utilization. How are the operations mapped onto MAC cells for winograd?
Sample h/w programming would help.

Thanks.

kchhavi, 2018-07-25:
Winograd allows passing 8 multiplications at a time to CACC. Further more winograd reduces the multiplies by over 2x for 3x3 convolution. Does this mean that winograd can improve the depth-wise convolution performance by nearly 16x (compared to direct)? Please confirm.

Thanks.

owenshe, 2018-08-01:
@kchhavi   How do you program NVDLA core for depth-wise convolution?

kchhavi, 2018-08-06:
@owenshe, For depth-wise convolution, I treated each depth (input feature-map) as a layer and chose the destination address for each depth such that they become contiguous in memory.

zhouweiscut, 2018-11-29:
@kchhavi, do you solve this utilization issue when do depth-wise conv on NVDLA ?

embedeepLHY, 2019-03-01:
Maybe you can try the FREE-TPU (https://github.com/embedeep/Free-TPU) which can do depth-wise convolution