nvdla/hw #178: How to build tests and run using Questa

mlhwvip, 2018-07-18:
Hello,

I do not have a licence for VCS. I would like to use Questa. Can you please help know if I can do that and how?

I am using nv_small from hw_master

Thanks,

fnormandin-xeda, 2018-10-03:
Hi @mlhwvip,

Please come at @xtreme-eda and take a look!

We currently have a version of the **nv_small** branch working with **Questa 10.4b**.

Please be aware that this project seems very sensitive to tool versions. We tested our drop using Questa 10.4b, SystemC2.3.0 (from Questa installation), gcc4.9.2, python3.6, no DesignWare.

This is still work in progress, but as of now we are able to:
- Execute "./tools/bin/tmake -build ready_for_test" without error, which includes:
  - build the RTL model and all VIPs
  - build the SystemC reference model
  - build the trace generator testbench
  - build the trace player testbench (including linking the SystemC model and DPI)
- Run a trace generator base test (cd verif/testbench/trace_generator; make run)
- Run a trace player base test (cd verif/testbench/trace_player; make run)

We are currently working on the **verif_protection** target.

We are also working on supporting **Cadence IUS**.

Cheers!

fnormandin-xeda, 2018-10-05:
Duplicate of #143