nvdla/hw #179: eperl error in file NV_NVDLA_CDMA_wt.v

ddkkevin, 2018-07-20:
In file https://github.com/nvdla/hw/blob/master/vmod/nvdla/cdma/NV_NVDLA_CDMA_wt.v,  "//: my $b0; //: my $b1; " should be added below line 1611,  otherwise vmod build will fail when define WEIGHT_DATA_TYPE_INT16_FP16 in spec.

mmaciag, 2018-07-20:
From what I know, HW has some pending important updates regarding _spec_ file "configurability" yet, so we could eventually use _nv_large_ or _nv_whatever_we_want_ (not only _nv_small_). I would appreciate to hear what is the ETA for this update by the way.

nvhook, 2018-07-24:
@ddkkevin Have you experienced this failure?

darrenleong, 2018-07-25:
i encountered this error also when try to build vmod using this definition (WEIGHT_DATA_TYPE_INT16_FP16)