nvdla/hw #185: Does nvdla_2

chifredhong, 2018-07-26:
I'm compiling the Virtual Simulator for debug version, when I  make to build the simlulator , it occurs the error below :
[ 74%] Building CXX object libs/greenlib/greenscript/lib/CMakeFiles/greenscript.dir/__/src/greenscript.cpp.o
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp: In constructor ‘gs::script::GreenScriptModule::GreenScriptModule(sc_core::sc_module_name, const char*)’:
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp:178:39: error: ‘PyString_FromString’ was not declared in this scope
   name_py = PyString_FromString(name());  // new ref
                                       ^
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp: In member function ‘void gs::script::GreenScriptModule::load(const char*)’:
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp:221:46: error: ‘PyString_AsString’ was not declared in this scope
       std::string s(PyString_AsString(path_py));
                                              ^
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp: In member function ‘void gs::script::GreenScriptModule::add_to_pythonpath(const char*)’:
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp:262:47: error: ‘PyString_FromString’ was not declared in this scope
   PyObject *path_py = PyString_FromString(path);  // new ref
                                               ^
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp: In member function ‘bool gs::script::GreenScriptModule::private_load(const char*)’:
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp:278:68: error: ‘PyFile_FromString’ was not declared in this scope
   PyObject* PyFileObject = PyFile_FromString((char *)fullname, mode);
                                                                    ^
In file included from /usr/include/python3.5m/Python.h:115:0,
                 from /home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp:38:
/home/chifred/NVDLA/my_compile/vp/libs/greenlib/greenscript/src/greenscript.cpp:279:56: error: ‘PyFile_AsFile’ was not declared in this scope
   PyObject *ret = PyRun_File(PyFile_AsFile(PyFileObject),
                                                        ^
/usr/include/python3.5m/pythonrun.h:160:23: note: in definition of macro ‘PyRun_File’
     PyRun_FileExFlags(fp, p, s, g, l, 0, NULL)
                       ^
libs/greenlib/greenscript/lib/CMakeFiles/greenscript.dir/build.make:62: recipe for target 'libs/greenlib/greenscript/lib/CMakeFiles/greenscript.dir/__/src/greenscript.cpp.o' failed
make[2]: *** [libs/greenlib/greenscript/lib/CMakeFiles/greenscript.dir/__/src/greenscript.cpp.o] Error 1
CMakeFiles/Makefile2:2079: recipe for target 'libs/greenlib/greenscript/lib/CMakeFiles/greenscript.dir/all' failed
make[1]: *** [libs/greenlib/greenscript/lib/CMakeFiles/greenscript.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

now my NVDLA HW VERSION is nvdla_2, I could compile successfully before I git pull new version. Is the nvdla_2 not supported  by simulator? or there is some other error I did't find?

anakin1028, 2018-07-30:
@chifredhong I meet the same problem, and changing Python3 to Python2 in the CMakeCache.txt file works for me.

mmaciag, 2018-07-31:
PyFile_FromString changed to PyBytes_FromString. Unfortunately python3 C-API has more such 'little' changes. One solution is editing CMakeCache.txt as @anakin1028 wrote, or you could try second solution from [here(https://github.com/opencog/opencog/issues/48#issuecomment-19075662)

chifredhong, 2018-07-31:
I install a "libpython2.7-dev" and solve the problem ,the issue is caused by my new installed python3.Thank you  anyway ~@anakin1028

chifredhong, 2018-07-31:
I have solve the issue by installing a lib. Thank your for your repl~ @mmaciag

closed 2018-07-31 by chifredhong