nvdla/hw #193: Floating point exception for VCS simulation on NVDLA Small

m1kezh, 2018-08-06:
Hi, After built the vmod and verif for NVDLA small, I tried to running single convolution test as verif guide mentioned : TOT/verif/tools/run_test.py -P nv_small dc_24x33x55_5x5x55x25_int8_0 -outdir dc_24x33x55_5x5x55x25_int8_output -wave -v nvdla_utb

An unexpected termination has occurred in ~/work/NVDLA/v2/hw-master/outdir/nv_small/verif/testbench/trace_player/simv due to a signal: Floating point exception

. It seems that VCS has encountered some internal error and collapsed. 

I upload the logs (testout) for this simulation collapsed.

I found that only if write CDMA D_OP_ENABLE register to '1' will cause simulator collapse.
reg_write(NVDLA_CDMA.D_OP_ENABLE_0, 0x1);

If I delete this line in dc_24x33x55_5x5x55x25_int8_0/dc_24x33x55_5x5x55x25_int8_0.cfg or change it to
reg_write(NVDLA_CDMA.D_OP_ENABLE_0, 0x0);  this collapse will not happen but finally  dc_24x33x55_5x5x55x25_int8_0 will get 'TEST FAILED' with UVM_FATAL :    1

Thanks!
Mike

Error log:

During SystemC/HDL delta-sync loop (RSysC) During execution of one SystemC delta cycle 
sc_time_stamp=32799500 sysc_new_sync=2 sysc_on_top=0 RSysC_cnt=85774 sysc_tli=1 
SC-process="SC_THREAD" name(SC)="nvdla_tb_top_rm_top_module_inst_nvdla_top_sc_adapter_inst.nvdla_top_sc_inst.nvdla_core.csc.DataLoadSequenceThread" parent(SC)="nvdla_tb_top_rm_top_module_inst_nvdla_top_sc_adapter_inst.nvdla_top_sc_inst.nvdla_core.csc" mhpi="nvdla_tb_top.rm_top_module_inst.nvdla_top_sc_adapter_inst.nvdla_top_sc_inst.nvdla_core.csc" 
Command line: ~/work/NVDLA/v2/hw-master/outdir/nv_small/verif/testbench/trace_player/simv -l testout +wave +UVM_MAX_QUIT_COUNT=1 +uvm_set_action=*,UVM/COMP/NAME,UVM_WARNING,UVM_NO_ACTION +uvm_set_action=*,UVM/RSRC/NOREGEX,UVM_WARNING,UVM_NO_ACTION +uvm_set_config_string=*,trace_file_path,~/work/NVDLA/v2/hw-master/health_exam/dc_24x33x55_5x5x55x25_int8_output/dc_24x33x55_5x5x55x25_int8_0/dc_24x33x55_5x5x55x25_int8_0.cfg

--- Stack trace follows:

Dumping VCS Annotated Stack:
#0  0x0000003239a0b5bc in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00002aaac1c6bbc0 in ndpBIMT_ConsumerMain () from /EDA/synopsys/verdi/Verdi_N-2017.12-SP2/share/PLI/lib/LINUXAMD64/libsscore_vcs201712.so
#2  0x0000003239a079d1 in start_thread () from /lib64/libpthread.so.0
#3  0x00000032396e89dd in clone () from /lib64/libc.so.6
#0  0x00000032396ac65d in waitpid () from /lib64/libc.so.6
#1  0x000000323963e609 in do_system () from /lib64/libc.so.6
#2  0x000000323963e940 in system () from /lib64/libc.so.6
#3  0x00002aaaac37a10b in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313e639d2928478c79b3f673f99203bcf8be64600612100082236bffb2007f1e0ef9 () from /EDA/synopsys/vcs-mx/N-2017.12-SP2/linux64/lib/liberrorinf.so
#4  0x00002aaaac37bb16 in SNPSle_10ee25eff68cd8461c9146fa1d0b35e87067f3c8015b313efba706aab251478fa49e66610e453774633a6c152e7ef778f2202cda681f3d4e () from /EDA/synopsys/vcs-mx/N-2017.12-SP2/linux64/lib/liberrorinf.so
#5  0x00002aaaac3748c5 in SNPSle_d35ca1ff70d465c2b9b1a72eee90a506fdd009d3de3db1de () from /EDA/synopsys/vcs-mx/N-2017.12-SP2/linux64/lib/liberrorinf.so
#6  0x00002aaaae87228f in SNPSle_64133461705005bb725549e2e6fa1b3f () from /EDA/synopsys/vcs-mx/N-2017.12-SP2/linux64/lib/libvcsnew.so
#7  0x00002aaaae6c575e in SNPSle_82244d58c54c18c70d63edc9becab634 () from /EDA/synopsys/vcs-mx/N-2017.12-SP2/linux64/lib/libvcsnew.so
#9  0x00002aaaaae1bcda in scsim::cmod::NV_NVDLA_csc::SendDataToMacSequencerDirectConvCommon() () from ~/work/NVDLA/v2/hw-master/outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/release/lib/libnvdla_cmod.so
#10 0x00002aaaaae1942a in scsim::cmod::NV_NVDLA_csc::DataLoadSequenceThread() () from ~/work/NVDLA/v2/hw-master/outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/release/lib/libnvdla_cmod.so
#11 0x00000000025bbf7c in sc_core::sc_process_b::semantics (this=0x436abb0) at vcs_cbug_step_into.cpp:80
#12 0x00000000025a6bcb in sc_core::sc_thread_cor_fn(void*) ()
#13 0x00000000025bc10c in sc_cor_qt_wrapper ()
#14 0x0000000000000000 in ?? ()

Process VmPeak: 941704 kb, VmSize: 903040 kb
System Free Memory: 11095020 kb, System Free Swap: 52428796 kb

killerzula, 2019-09-09:
Hey, can you tell me how did you get that TOT/verif/tools folder? I built the testbench with the command
`./tools/bin/tmake -build ready_for_test`
but the tools folder inside verif is not there. Kindly help me. Thanks a lot.

wxcun121, 2024-10-11:
do you have  fixed it ?

shazib-summar, 2024-10-11:
@wxcun121 its been 5 years now buddy. I've moved on to better and less crazy things.

LeiWang1999, 2024-10-11:
@shazib-summar lol