nvdla/hw #194: Can VLIB_BYPASS_POWER_CG macro be valid for FPGA?

rainandy, 2018-08-06:
I want to implement nv_small on FPGA, but I found that defining VLIB_BYPASS_POWER_CG macro will make simulation wrong. Can anybody explain that? thanks!

xivisi, 2018-08-06:
if not define VLIB_BYPASS_POWER_CG, the gate clocks make bad THW,TLW in fpga.

mmaciag, 2018-08-06:
In FPGA you have very little resources for physical clock gating. See detailed answer [here](https://forums.xilinx.com/t5/General-Technical-Discussion/Reg-Clock-gating-for-FPGAs-good-or-bad/td-p/279154). This is answer specific to Xilinx products, but it's essence is valid for other vendors.

liuhengibm, 2018-08-16:
To disable clock gate, you can try to define 
FIFOGEN_MASTER_CLK_GATING_DISABLED, 
RAM_DISABLE_POWER_GATING_FPGA, 
VLIB_BYPASS_POWER_CG, 
NV_FPGA_FIFOGEN, 
FPGA 
at the same time.