nvdla/hw #195: Error when build ready_for_test

ptzheng, 2018-08-09:
Hello, after following the build tree instructions for the Nv_small configuration, but I got an error when build test bench using commands:
`./tools/bin/tmake -build ready_for_test`

The output is as follow:
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in verif/testbench/trace_generator 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in vmod/vlibs 
[TMAKE]: building nv_small in vmod/include 
[TMAKE]: building nv_small in vmod/rams/model 
[TMAKE]: building nv_small in vmod/rams/synth 
[TMAKE]: building nv_small in vmod/rams/fpga/small_rams 
[TMAKE]: building nv_small in vmod/nvdla/apb2csb 
[TMAKE]: building nv_small in vmod/nvdla/cdma 
[TMAKE]: building nv_small in vmod/nvdla/cbuf 
[TMAKE]: building nv_small in vmod/nvdla/csc 
[TMAKE]: building nv_small in vmod/nvdla/cmac 
[TMAKE]: building nv_small in vmod/nvdla/cacc 
[TMAKE]: building nv_small in vmod/nvdla/sdp 
[TMAKE]: building nv_small in vmod/nvdla/pdp 
[TMAKE]: building nv_small in vmod/nvdla/cfgrom 
[TMAKE]: building nv_small in vmod/nvdla/cdp 
[TMAKE]: building nv_small in vmod/nvdla/bdma 
[TMAKE]: building nv_small in vmod/nvdla/rubik 
[TMAKE]: building nv_small in vmod/nvdla/car 
[TMAKE]: building nv_small in vmod/nvdla/glb 
[TMAKE]: building nv_small in vmod/nvdla/csb_master 
[TMAKE]: building nv_small in vmod/nvdla/nocif 
[TMAKE]: building nv_small in vmod/nvdla/retiming 
[TMAKE]: building nv_small in vmod/nvdla/top 
[TMAKE]: building nv_small in verif/vip/reference_model/nvdla_cmod_wrap 
Makefile:34: warning: overriding commands for target `../../../../outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
../../../../cmod/Makefile:162: warning: ignoring old commands for target `../../../../outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/libnvdla_cmod.so'
../../../../cmod/glb/gen/gec_reg_model.cpp: In member function ‘void scsim::cmod::gec_reg_model::GecUpdateVariables(scsim::cmod::CNVDLA_GEC_REGSET*)’:
../../../../cmod/glb/gen/gec_reg_model.cpp:148:90: error: ‘class scsim::cmod::CNVDLA_GEC_ERRSLICE0_MISSIONERR_STATUS’ has no member named ‘uERR17’
         errslice0_missionerr_status_err17_ = reg_group_ptr->rERRSLICE0_MISSIONERR_STATUS.uERR17();
                                                                                          ^
make: *** [../../../../outdir/nv_small/verif/vip/reference_model/nvdla_cmod_wrap/glb/gen/gec_reg_model.ref.o] Error 1
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

My tree.make is as follow:
USE_DESIGNWARE  := 0
DESIGNWARE_DIR  := /home/tools/synopsys/syn_2011.09/dw/sim_ver
CPP  := ~/gcc-4.8.2/bin/cpp
GCC  := ~/gcc-4.8.2/bin/gcc
CXX  := ~/gcc-4.8.2/bin/g++
PERL := ~/perl5.22.1/bin/perl
JAVA := ~/Downloads/jdk-10.0.2/bin/java
SYSTEMC := ~/systemc-2.3.0/
PYTHON := ~/anaconda3/bin/python
VCS_HOME := /apps/EDAs/Synopsys/vcs-mx/K-2015.09-SP2-1
NOVAS_HOME := /apps/EDAs/Synopsys/verdi/Verdi_M-2017.03-SP2
VERDI_HOME := /apps/EDAs/Synopsys/verdi/Verdi_M-2017.03-SP2
VERILATOR := /usr/local/bin/verilator
CLANG := /usr/bin/clang

Can anyone give me some ideas? Thanks in advance

piotr-aldec, 2019-04-12:
Hello,

Were you able to solve that issue?
I had similar but with uErr1

Best Regards,
Piotr

xyy-casia, 2020-04-22:
> Hello,
> 
> Were you able to solve that issue?
> I had similar but with uErr1
> 
> Best Regards,
> Piotr

Hello! Have you sorted this problem out?
I met the same as yours exactly with uErr1.

Thanks in advance!

beckerasicskills, 2022-09-15:
Any progress of any kind?
I have monitored those "verif/vip/reference_model/nvdla_cmod_wrap" build issue for nv_small regarding the unix path where you actually cloned the repo.
It gives me different results depending on how many level of hierarchy I have from root file system