nvdla/hw #196: Implementing NVDLA on FPGA

nookfoo, 2018-08-09:
Hi,
my goal is to eventually run a configurable nv_small on Zynq Ultrascale+ while using an nvdc compiled caffe model for inference. I am still a novice in the field of FPGAs, NNs, etc. and have some trouble comprehending the basic implementation flow of NVDLA. I have read through documentation and a few related discussions (#110, nvdla/sw#70) and would like to roughly reiterate what I have gathered so far.

General steps for implementation of NVDLA on Zynq Ultrascale+:

1. Generate RTL model of nv_small using [hw manual](http://nvdla.org/hw/v2/environment_setup_guide.html)
    - try different configurations of nv_small 

2. Synthesis and Implementation of RTL model for Zynq Ultrascale+ (using Vivado in this case)
    - setup connections, RAMs etc.
    - consider [sub-steps ](https://github.com/nvdla/hw/issues/110#issuecomment-377676962)

3. Setup nvdla_runtime (umd, kmd) on Linux distro running on Zynq Ultrascale+ SoC 
    - refer to this [comment](https://github.com/nvdla/hw/issues/110#issuecomment-394312879)
    - run sanity checks etc. ([test applications](http://nvdla.org/sw/test_application.html))
    - use prebuilts/linux provided by nvdla/sw
    - use provided loadables or generate self-configured NN with nvdla_compiler

I'm aware this is a very rough estimate on what needs to be done and there will a multitude of sub-steps and issues that will need to be resolved, but I would greatly appreciate feedback on if this is the right approach. Thanks!

mmaciag, 2018-08-10:
Yeah, these additional minor steps are hard to define but almost inevitable, but in general the plan sounds good to me. Except for:

> generate self-configured NN with nvdla_compiler

This will be eventually possible if nvdla_compiler gets support for configurations other than nv_full (a.k.a. [nvdlav1](https://github.com/nvdla/hw/tree/nvdlav1)). Unfortunately the official [roadmap](http://nvdla.org/roadmap.html) is out-dated and nobody knows when it may happen.

xiaoguoer, 2018-10-30:
@nookfoo 
For your second step, could you please share how do you do with "Add DW02_tree DW_lsb DW_minmax files" and "global define"  in the [sub-steps](https://github.com/nvdla/hw/issues/110#issuecomment-377676962) you refered??

Thanks in advance for any reply I might have.

nookfoo, 2018-10-31:
Hi, "Add DW02_tree DW_lsb DW_minmax files" was not necessary for me since that is used for DesignWare products and simulation I think. I used Vivado for my implementation.

I am myself not sure yet where to add global defines. For now I have them added under Project Settings -> Verilog Options in Vivado and also in NV_HVACC_NVDLA_tick_defines.h.

xiaoguoer, 2018-10-31:
@nookfoo thank you ~
[the answer](https://github.com/nvdla/hw/issues/246#issuecomment-434579257) from mmaciag for this question. hope helpful for you.

smartchen2k, 2018-11-01:
you can include the path of  *defines.h in the "simulation setting" in vivado.
And if you want to make some macro globle, you just right click the file and "set as globle include".

smartchen2k, 2018-11-01:
So up to now, there is no specific directions for which directive macros should be defined if you implement the master branch nv_small into FPGA. For those who has done the simulation and synthesis, they can only check if their settings are right through runing test. 
Am I right? 
So, sharing the setting of the successful project and your configurations are valued much for others.

mmaciag, 2019-01-15:
@peterzh2018888
You need to make a custom IP core wrapper around the NVDLA. Only then you can use it as a component in your block design. The signals quite easily map to AXI4 and APB. Have you seen this thread? https://github.com/nvdla/hw/issues/110.

peterzh2018, 2019-01-16:
#110. Not specific enough to operate and implement. Can you be practical? It's not simple at all.

peterzh2018, 2019-01-16:
> @peterzh2018888
> You need to make a custom IP core wrapper around the NVDLA. Only then you can use it as a component in your block design. The signals quite easily map to AXI4 and APB. Have you seen this thread? #110.
Can you be practical? It's not simple at all.#110. Not specific enough to operate and implement.

peterzh2018888, 2019-02-25:
Test on ZCU102 Board.

= Run NN/NN_L0_1_small_fbuf
creating new runtime context...
Emulator starting
submitting tasks...
[ 2210.896668] Enter:dla_read_network_config
[ 2210.903672] Exit:dla_read_network_config status=0
[ 2210.909643] Enter: dla_initiate_processors
[ 2210.915004] Enter: dla_submit_operation
[ 2210.920099] Prepare Convolution operation index 0 ROI 0 dep_count 1
[ 2210.927667] Enter: dla_prepare_operation
[ 2210.932896] processor:Convolution group:1, rdma_group:0 available
[ 2210.940346] Enter: dla_read_config
[ 2210.945125] Exit: dla_read_config
[ 2210.949831] Exit: dla_prepare_operation status=0
[ 2210.955870] Enter: dla_program_operation
[ 2210.961236] Program Convolution operation index 0 ROI 0 Group[1]
[ 2210.968764] no desc get due to index==-1
[ 2210.974179] no desc get due to index==-1
[ 2210.979577] no desc get due to index==-1
[ 2210.984954] no desc get due to index==-1
[ 2210.990313] no desc get due to index==-1
[ 2210.995644] Enter: dla_op_programmed
[ 2211.000630] Update dependency operation index 2 ROI 0 DEP_COUNT=2
[ 2211.008178] Update dependency operation index 1 ROI 0 DEP_COUNT=1
[ 2211.015723] enable SDP in dla_update_dependency as depdency are resolved
[ 2211.023905] Enter: dla_enable_operation
[ 2211.029220] exit dla_enable_operation without actual enable due to processor hasn't been programmed
[ 2211.039839] Exit: dla_enable_operation status=0
[ 2211.045950] Exit: dla_op_programmed
[ 2211.050997] Exit: dla_program_operation status=0
[ 2211.057176] Exit: dla_submit_operation
[ 2211.062484] Enter: dla_dequeue_operation
[ 2211.067953] Dequeue op from Convolution processor, index=2 ROI=0
[ 2211.075536] Enter: dla_submit_operation
[ 2211.080938] Prepare Convolution operation index 2 ROI 0 dep_count 1
[ 2211.088792] Enter: dla_prepare_operation
[ 2211.094292] processor:Convolution group:0, rdma_group:0 available
[ 2211.101991] Enter: dla_read_config
[ 2211.106995] Exit: dla_read_config
[ 2211.111900] Exit: dla_prepare_operation status=0
[ 2211.118105] Enter: dla_program_operation
[ 2211.123600] Program Convolution operation index 2 ROI 0 Group[0]
[ 2211.131226] no desc get due to index==-1
[ 2211.136737] no desc get due to index==-1
[ 2211.142218] no desc get due to index==-1
[ 2211.147668] no desc get due to index==-1
[ 2211.153083] no desc get due to index==-1
[ 2211.158471] Enter: dla_op_programmed
[ 2211.163499] Update dependency operation index 6 ROI 0 DEP_COUNT=3
[ 2211.171072] Update dependency operation index 3 ROI 0 DEP_COUNT=2
[ 2211.178616] Exit: dla_op_programmed
[ 2211.183552] Exit: dla_program_operation status=0
[ 2211.189625] Exit: dla_submit_operation
[ 2211.194807] Exit: dla_dequeue_operation
[ 2211.200064] Enter: dla_submit_operation
[ 2211.205327] Prepare SDP operation index 1 ROI 0 dep_count 0
[ 2211.212329] Enter: dla_prepare_operation
[ 2211.217670] processor:SDP group:0, rdma_group:1 available
[ 2211.224484] Enter: dla_read_config
[ 2211.229290] Exit: dla_read_config
[ 2211.233999] Exit: dla_prepare_operation status=0
[ 2211.240023] Enter: dla_program_operation
[ 2211.245346] Program SDP operation index 1 ROI 0 Group[0]
[ 2211.252076] no desc get due to index==-1
[ 2211.257384] no desc get due to index==-1
[ 2211.262694] no desc get due to index==-1
[ 2211.268009] no desc get due to index==-1
[ 2211.273323] Enter: dla_op_programmed
[... 134 lines cut ...]
[ 2212.042876] Exit:dla_handle_events, ret:0
[ 2212.048188] Enter:dla_handle_events, processor:SDP
[ 2212.054317] Exit:dla_handle_events, ret:0
[ 2212.059688] Enter:dla_handle_events, processor:PDP
[ 2212.065855] Exit:dla_handle_events, ret:0
[ 2212.071217] Enter:dla_handle_events, processor:CDP
[ 2212.077357] Exit:dla_handle_events, ret:0
[ 2212.082717] Enter:dla_handle_events, processor:RUBIK
[ 2212.089054] Exit:dla_handle_events, ret:0
[ 2212.094454] Enter:dla_handle_events, processor:BDMA
[ 2212.100733] Exit:dla_handle_events, ret:0
[ 2212.106130] Enter:dla_handle_events, processor:Convolution
[ 2212.113015] Exit:dla_handle_events, ret:0
[ 2212.118409] Enter:dla_handle_events, processor:SDP
[ 2212.124584] Exit:dla_handle_events, ret:0
[ 2212.129985] Enter:dla_handle_events, processor:PDP
[ 2212.136181] Exit:dla_handle_events, ret:0
[ 2212.141594] Enter:dla_handle_events, processor:CDP
[ 2212.147788] Exit:dla_handle_events, ret:0
[ 2212.153183] Enter:dla_handle_events, processor:RUBIK
[ 2212.159561] Exit:dla_handle_events, ret:0

Dead here, can't keep running.

What could be the cause of this problem? Has anyone ever had the same problem? How to fix it.

embedeepLHY, 2019-03-01:
If you just looking for a TPU design with FPGA, maybe the "FREE-TPU" (https://github.com/embedeep/Free-TPU) is suit for you

Okaymaddy, 2019-06-04:
I am trying to create the nvdla IP on Vivado and I am confused on how to do the same. If anyone can provide the steps it will be really helpful.