nvdla/hw #197: How the input.txn file is generated.

gowthar, 2018-08-10:
Is there any script file which is generating input.txn and input_feature_map.dat and input_weight.dat ?

wxbbuaa2011, 2018-09-26:
also interested in this

wxbbuaa2011, 2018-09-26:
@debjitpal Hello, have you solved your question? I hope to followyour work.

wxbbuaa2011, 2018-09-26:
@gowthar Hello, have you solved your question? I hope to followyour work.

wxbbuaa2011, 2018-09-26:
@DC11011100 Can you give me this file? Thank you very much.

wxbbuaa2011, 2018-10-07:
Can you give me this file?

DC11011100, 2019-03-08:
@wxbbuaa2011 
Sorry I no longer work on this stuff and don't plan on picking it back up for sometime. My plan and where I left off:

Last I checked they didn't provide the hardware loader source code but they did offer a binary for their emulated machine, _which also runs a kernel you can alter_. So I modified the kernel at the hardware abstraction layer to dump logs to some log in a shared memory space on the emu-machine memory which I could later access offline on the machine running the emu-mach.

The dumped sequence of hardware transactions can probably be used to compile *.txn, while cross-checking syntax with that shown in their provided example *.txn files, and remapping memory addresses to the configuration you're going to program.

Since the memory space depends on the HW config you must determine dependent sub address spaces to whatever HW configuration you're emu-mach is running (i.e. nv-small, nv-large, etc.) and the kind the nvloader binary is expecting. This sort of info, while finicky to find, is nonetheless described in their provided HW docs.

Good luck!