nvdla/hw #199: Has anyone succeeded running on SoC-FPGA?

xivisi, 2018-08-14:
I have built connected nv_small(7cf6ad5) to A53 hard arm cpu, but  not ran well.
So I captured 371  commands from apb(apb2csb) bus, and repeated them in rtl-similation environment。
Some output signals of nv_small top module were unsettling, like axi id, data, last, etc.

Is there somebody, can solve this problem?

mmaciag, 2018-08-14:
Hi, please have a look on these two threads: nvdla/hw#110, #70. Basing on them perhaps you will be able to extend your check list.

xivisi, 2018-08-17:
@mmaciag  I have tried but  have no idea how to solve these problems.

mmaciag, 2018-08-17:
Hard to tell not knowing the connections with host CPU. If you observe invalid outputs then maybe you have incorrectly applied reset signal. Note that reset is active _low_.

ptzheng, 2018-08-21:
@mmaciag Hi, I am confused about _dla_reset_rstn_ and _direct_reset_ . Could you please tell me about the relationship between these two reset signals? If I want to run on fpga, how can i connect these two signals to the board?

xivisi, 2018-08-31:
@mmaciag  It still does't work.
In  rtl similation:
    1, config 
        A, git version: Latest commit 7cf6ad5 
        B, selfdefined testbench
            a), xilinx block ram relpace nv_rams
                nv_ram_rws* no input output register;
                nv_ram_rwsp* have input output register;
                nv_ram_rwst* only input register;
                nv_ram_rwsthp* both io registers and output bypass;
            b),  xilinx axi block ram to ddb interface
                64bit, 2M capacity;
                high address bit ignor e
            c), NV_NVDLA_apb2csb as bridge between selfdefined apb master and csb;
        C, global defined: 
            DISABLE_TESTPOINTS,
            SYNTHESIS,
            FPGA,
            //NV_SYNTHESIS (no real effect, after searched all of vmod output source code)
            //RAM_INTERFACE(no effect, after replace all nv_rams)
            VLIB_BYPASS_POWER_CG,
            FIFOGEN_MASTER_CLK_GATING_DISABLED,
            RAM_DISABLE_POWER_GATING_FPGA
        D, interface config
            a), dla_core_clk and dla_csb_clk, same clock;
            b), global_clk_ovr_on, tmc2slcg_disable_clock_gating, test_mode, constant 1'B0;
            c), dla_reset_rstn, direct_reset_, same low level reset.
        
    2, working phenomenon
        A, so many "X" signal, when write all csb code, nothing happened.
        B, add initial code for no reset inner register, it seems so well.
            trace test "cdp_8x8x32_lrn3_int8_1"
            after csb write(addr:c008, value:0000_0001), ddb axi read datas, from 0000_0000 to 0000_0440,
            after axi block ram response, nvdla no other action.

xivisi, 2018-09-03:
@mmaciag  Yes, NV_FPGA_FIFOGEN. but add it, still do not work well.
I can not solve these probleams, now. go on doing.

xivisi, 2018-09-03:
@mmaciag  could you tell me you configuration?

huangwei858, 2019-03-13:
> I think `test_mode = 0` effectively disables `direct_reset_` path and it can be tied to constant 1'b0 too. I see also `NV_FPGA_FIFOGEN` in my definition `.vh` file - grep is pointing csb_master directory so it might be significant. I cannot remember what kind of issue I had that caused me to use `include clause instead of global macro definition. And I like git editor :)

Hey, I'm glad to happy when get this comments. Thank you for your efforts in nv_small. I've passed synthesis successfully through your guide. But I get some error about the IOB is too large(125%) when synthesis completely. Could you help me to solve this problem?