nvdla/hw #201: Can anyone give me some advice porting NVDLA into Intel FPGA?

j0210003, 2018-08-16:
Hello, I'm trying to port nv_full version of NVDLA to Intel Arria 10 SX version (size : 660).

I've read some FPGA porting issues and found the same problems. (they seems to use Xilinx FPGA)
I could do synthesis but DSP usage is very poor for MACs due to the structures.
So, currently the FPGA is too small for the nv_full version.(combinational logic)

I'm trying to change the MAC structure for DSP porting to get through this.
Is there anyone already has done this work? Or is there any guide to do this?

Thank you in advance.

sun3388, 2018-08-16:
* Why not try nv_small firstly?
* nv_large is already released, this one is more preferred than nv_full if you want a larger one.

mmaciag, 2018-08-16:
@j0210003 I didn't work with nv_full, but this is going to be really expensive - the reports from #110 show it would fit only the largest Virtex VU440 from all Xilinx devices. The nv_full utilizes >80% resources of this _massive_ device. I am afraid you might have trouble fitting event the largest Arria device.

On the other side, _nv_small_ fits relatively small devices. Your SX 660 could maybe fit nv_medium, but you need to check.

@sun3388 I think current 'nv_large' does not support floating point, or does it?

sun3388, 2018-08-16:
yes, I think all scalable releases don't support FP for now.

j0210003, 2018-08-20:
@sun3388 Thanx for your comment. But from #196, I have read this - This will be eventually possible if nvdla_compiler gets support for configurations other than nv_full (a.k.a. nvdlav1). Unfortunately the official roadmap is out-dated and nobody knows when it may happen.

So, I thought currently there is no SW support for nv_small nor nv_large. Is it correct?

j0210003, 2018-08-20:
@mmaciag Hello, I don't have any manpower doing SW development. So, I want to use SW from nvdla. If I port nv_small, is it correct I can't use released SW? Did I correctly understand from your comment in #196?

nookfoo, 2018-08-22:
@j0210003 I think what is meant in that comment is that nv_small (currently under the branch hw-master) cannot be used for custom NN-models yet, but only for verification of provided loadables for now. 
The released sw contains drivers, loadables and sanity tests for an nv_small port.

j0210003, 2018-08-23:
@nookfoo Thank you for your opinion. I want to use it for acceleration of our custom NN, so I wanted to check it clearly. Then I have to use nv_full version anyway.