nvdla/hw #202: \perf\NVDLA_OpenSource_Performance.xlsx

wxbbuaa2011, 2018-08-17:
We can set the parameters in the file (NVDLA_OpenSource_Performance.xlsx) to calculate the time period of the MAC calculation; but there is a gap between the data I simulated? Is it because the MAC cycle calculated in this file does not calculate the time of CDMA and CBUF and CSC? Can the data generated by this file be used directly as a reference?

SCLUO, 2018-08-17:
From my experiences,  you may check and revise the Winograd option, the cycle count formula did not refer to "Winograd term" but directly to the "kernel size".  Winogard is not supported in small versions.

Also, current calculation does not consider the data movement time to CBUF. However, this data loading time can be neglected if you are running a "typical"  direct convolution layer,  which has a small IFM, and a medium kernel size.  NVDLA DMAs can cover the weight data movement under typical CONV calculation.  In opposite, when estimating some large IFM or large kernel sizes,  data movement time may not reflect.

wxbbuaa2011, 2018-08-17:
Thank you very much!

wxbbuaa2011, 2018-10-22:
The instruction stream obtained by vp, the number of cycles obtained by VCS simulation of the alexnet network is very different from the data in the excel table. In the same configure, my way is 2093882.0771cycles,and excel is the 895443.563.So which is right?
--

SCLUO, 2018-10-22:
Excel estimator does NOT include cycles of data movement, such as register configuring, initial CONV buffer filling.  Another two  issues need to be aware as well. (1) this estimator is based on nv_full, 16-bit inference, not useful for nv_small or 8-bit inference. (2) this estimator didn't include the CONV buffer full situation when your weight is large.  

Many stories if you would like to dive deep.