nvdla/hw #203: make: *** [sysc_build] Error 1

fcyer, 2018-08-17:
nv_small mode, when build ready_for_test generate errors below:
@localhost hw_small]$ ./tools/bin/tmake -build ready_for_test
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in verif/testbench/trace_generator 
[TMAKE]: building nv_small in spec/odif 
[TMAKE]: building nv_small in vmod/vlibs 
[TMAKE]: building nv_small in vmod/include 
[TMAKE]: building nv_small in vmod/rams/model 
[TMAKE]: building nv_small in vmod/rams/synth 
[TMAKE]: building nv_small in vmod/rams/fpga/small_rams 
[TMAKE]: building nv_small in vmod/nvdla/apb2csb 
[TMAKE]: building nv_small in vmod/nvdla/cdma 
[TMAKE]: building nv_small in vmod/nvdla/cbuf 
[TMAKE]: building nv_small in vmod/nvdla/csc 
[TMAKE]: building nv_small in vmod/nvdla/cmac 
[TMAKE]: building nv_small in vmod/nvdla/cacc 
[TMAKE]: building nv_small in vmod/nvdla/sdp 
[TMAKE]: building nv_small in vmod/nvdla/pdp 
[TMAKE]: building nv_small in vmod/nvdla/cfgrom 
[TMAKE]: building nv_small in vmod/nvdla/cdp 
[TMAKE]: building nv_small in vmod/nvdla/bdma 
[TMAKE]: building nv_small in vmod/nvdla/rubik 
[TMAKE]: building nv_small in vmod/nvdla/car 
[TMAKE]: building nv_small in vmod/nvdla/glb 
[TMAKE]: building nv_small in vmod/nvdla/csb_master 
[TMAKE]: building nv_small in vmod/nvdla/nocif 
[TMAKE]: building nv_small in vmod/nvdla/retiming 
[TMAKE]: building nv_small in vmod/nvdla/top 
[TMAKE]: building nv_small in verif/vip/reference_model/nvdla_cmod_wrap 
[TMAKE]: building nv_small in verif/testbench/trace_player 
make: *** [sysc_build] Error 1
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE

Anyone get the same error can help me, Thank you .

mmaciag, 2018-08-17:
In outdir/nv_small you will find bunch of log files, you should find more details on error there.

fcyer, 2018-08-20:
Yes,  I found it.

killerzula, 2019-09-19:
What did you find?