nvdla/hw #209: nvdla_tb_sim_cfg.py SyntaxError

kouzhentao, 2018-08-28:
Hi,

When ./verif/tools/run_test.py -P nv_small dc_24x33x55_5x5x55x25_int8_0 -outdir dc_24x33x55_5x5x55x25_int8_output -wave -v nvdla_utb

Traceback (most recent call last):
  File "./verif/tools/run_test.py", line 11, in <module>
    import nvdla_tb_sim_cfg
  File "/home/usr/kouzhentao/nvdla/hw/verif/tools/nvdla_tb_sim_cfg.py", line 221
    SB_CFG = { **CDMA_SB_CFG, 
                ^
SyntaxError: invalid syntax

thanks.

ic7x24, 2018-10-26:
please check your python version. NVDLA `master` branch requires Python3 in your $PATH.

xldeng-chn, 2021-04-07:
> 
> 
> Hi,
> 
> When ./verif/tools/run_test.py -P nv_small dc_24x33x55_5x5x55x25_int8_0 -outdir dc_24x33x55_5x5x55x25_int8_output -wave -v nvdla_utb
> 
> Traceback (most recent call last):
> File "./verif/tools/run_test.py", line 11, in
> import nvdla_tb_sim_cfg
> File "/home/usr/kouzhentao/nvdla/hw/verif/tools/nvdla_tb_sim_cfg.py", line 221
> SB_CFG = { **CDMA_SB_CFG,
> ^
> SyntaxError: invalid syntax
> 
> thanks.

Have you solved this problem? I also met this problem now. I don't know what 's wrong. I have already included python3.6 in my $PATH and tree.make python path.

[image]

[image]

but the problem still exists
[image]