nvdla/hw #210: MAC Cycle Calculation

kent820518, 2018-08-29:
Does anyone know what is the meaning of Ceiling(Ifmap Channel * Stride_H * Stride_W,16) in MAC Cycle Calculation of NVDLA Performance xlsx?

mmaciag, 2018-09-03:
I think this is related with [Atomic Operation](http://nvdla.org/hw/v1/ias/unit_description.html#atomic-operation) and the way how weight data is fetched from memory - regardless of MAC usage during Stripe Operation, 16 cycles are needed to load weight data.

Erdbeerenjanzi, 2019-05-13:
I think the whole formula does not work out.
Just made this minimum example and got some strange results:

input_height, input_width = 4, 4
filter_height, filter_width = 1, 1
input_feature_maps = 1
output_feature_maps = 16
stride_horizontal, stride_vertical = 1, 1
zero_padding = true

Which results in MAC_cycles = 2

If loading the weights was included in the formula, we should at least have 16 MAC_cycles.
The output image is 4x4 and should therefore need at least 16 cycles if channel-post-extension is not activated.

There are some further inconstistences and open questions like:
-What about image mode? 
-What about channel-post-extension?
-What about the depth and number of MAC cells (MAC_ATOMIC_C_SIZE_64, MAC_ATOMIC_K_SIZE_32)?
-According to the formula the MAC cycles scale reciprocally to the MAC number. I really doubt that since the utilization depends on the MAC number.