nvdla/hw #218: error in ./tools/bin/tmake -build verif_protection  (Master branch)

koroyuki, 2018-09-17:
Dear sir,
I am working on master branch. 
 ./tools/bin/tmake -build vmod and ./tools/bin/tmake -verbose ready_for_test went well.
However,  ./tools/bin/tmake -build verif_protection  ends up with SyntaxError as shown below.
I am using Python 3.5.2 and perl v5.16.3. colorama python package is installed.
Does anybody you give me a suggestion?

[tago@aplab5 nvdla_hw_master-apl]$ ./tools/bin/tmake -build verif_protection
[TMAKE]: building nv_small in spec/defs 
[TMAKE]: building nv_small in spec/manual 
[TMAKE]: building nv_small in verif/testbench/trace_generator 
---- omitted. no errors ----
[TMAKE]: building nv_small in vmod/nvdla/top 
[TMAKE]: building nv_small in verif/vip/reference_model/nvdla_cmod_wrap 
[TMAKE]: building nv_small in verif/testbench/trace_player 
[TMAKE]: building nv_small in verif/protection 
Traceback (most recent call last):
  File "/autohome/ozuser/APD2/tago/nvdla_hw_master-apl/verif/tools/run_plan.py", line 16, in <module>
    from run_report  import RunReport
  File "/autohome/ozuser/APD2/tago/nvdla_hw_master-apl/verif/tools/run_report.py", line 290
    print(f'[INFO] Will update regression status after {colorama.Fore.RED}%0d{colorama.Style.RESET_ALL} seconds later ...' % val, end='\r')
                                                                                                                         ^
SyntaxError: invalid syntax
make: *** [run] Error 1
Died at ./tools/bin/tmake line 254.
[TMAKE]: DONE
[tago@aplab5 nvdla_hw_master-apl]$ 
=====================================================

GuyEichler, 2019-03-02:
I'm having the same error. Did you find a solution?

peter-d, 2019-03-21:
I had the same issue. This is because the `f'...'` syntax for strings is supported from python 3.6 only. I had this error when running with python 3.5.1 but not with 3.6 anymore.

koroyuki, 2019-03-22:
Dear Peter,
It is a great help to me.  I almost give up the NVDLA execution. I try
again with Python 3.6.

Thanks again
Tago.