nvdla/hw #221: What's the intention of "TOT > ./tools/bin/tmake -build vmod"

anakin1028, 2018-09-20:
Hi all,

Follows the instructions given http://nvdla.org/hw/v2/environment_setup_guide.html
I can build the RTL successfully. But when I check the outdir/nv_small/vmod directories there are two types extension files. One is *.v and another one is *.v.vcp and the contents are the same. 

The *.v files are already in hw/vmod directory, so why we want to do this command again to generate the same things?

Any guidance is appreciated, thanks!

mmaciag, 2018-09-20:
That's NVIDIA Magic Build System (TM). 

Just kidding. The build scripts are actually quite smart way to handle various architecture settings defined in _spec/defs_ directory.

These .v.vcp files are intermediate files created during passing _$TOT/vmod_ Verilog files through:
* Magic Perl script that transforms illegal characters (_for C preprocessor_) to something else.
* GCC preprocessor (_cpp_), which substitutes C-style preprocessor stuff (`#ifdef`, `#ifndef`, etc.)
* Magic Perl script that transforms 'something else' back to illegal characters (_for C preprocessor_).
* Another magic Perl script, which reads inline comments like `//:` and transforms it into Verilog code, dependent on architecture settings in _.spec_ files).

The _.v.vcp_ are temporary files, and after all they could be removed by the build system. But they aren't.

Hope it helps!

mmaciag, 2018-09-20:
Yes, the synthesisable source code is under outdir (the .v files and few .vh include files).

Well, I answered the 'how' question. I guess the 'why' questions should be answered by the authors :). 

The reasoning for `//:` stuff could be easier generation of tedious code like flip-flop instantiation. Or make the source code more concise. On the other hand I find such code quite hard to follow.

I see one reason why they might need to use C preprocessor: with Verilog preprocessor they couldn't substitute macros that appear in `//:` comments.

anakin1028, 2018-11-22:
@mmaciag  thanks!

closed 2018-11-22 by anakin1028